J2Kbr/TitanTwo #155: SD-card problem after coding mistake with pmem_save

Scachi, 2018-07-28:
I got the problem back again where I couldn't connect to the T2 with GTuner with the SD-card already inserted into T2.

Start the T2 without sd-card, Start GTuner IV, insert the SD-card, everything (all slot content) shows and the T2 stays connected.

With the SD-card inserted and the T2 running and starting GTuner IV my T2 gets disconnected (error 2)

The last thing I have done before my T2 stopped connecting with the SD-card inserted was rewriting some code that was writing every second to pmem in "slot 9" out of mistake.
I was using the slot 9 for some minutes trying to find the cause until I realized I had the timer check if the pmem should be saved or not wrong.
I was using a "if ( Now=1 )" instead of "if (Now==1 )" so it did call every second "pmem_save()" causing lag when running "slot 9".

During the search for the cause I closed GTuner and the problem with the SD-card occurred.

Deleting the files from slot 9 on the sd-card didn't fix it.
Formating the sd card did fix the issue.

Maybe you can reproduce it and fix the sd-card problem or have an idea if this could be the cause at all ?

This is  my header file with the mistake I was using in a script
in function "void PM_Update(uint16 now) " :
```
#ifndef _ICPM_UPDATE
#define _ICPM_UPDATE
  
// update the Interactive Configuration window and pmem_save only every n secs
// use anywhere in your code with a toggle when you want to update
// for pmem_save execution flag:
//     PM_Update(TRUE);
//
// use for interactive windows refreshing flag:
//    IC_Update(TRUE);
//
// for both use: ICPM_Update(TRUE);
// to change the intervall/speed it is allows to updates use the time in ms like: ICPM_Update(3000);
// ! The update interval has to be >= 1000ms or it will have no effect

uint16 ICPM_Interval = 1000;
  
main {
  IC_Update(); // only run once when it was called once with ICRefresh(TRUE); somewhere else
  PM_Update(); // only run once when it was called once with PM_Update(TRUE); somewhere else
}

// update both
void ICPM_Update(uint16 now) {
  if ( now >= 1000 ) ICPM_Interval=now;
  IC_Update(now);
  PM_Update(now);
}

// Update the interactive configuration window, but not faster than every 1 second
void IC_Update(uint16 now) {
  static bool ICUpdate = FALSE;
  static uint32 ICULast = system_time();
  if (now==1) ICUpdate=now;
	if (ICUpdate && ICULast < system_time()) {
		printf("IC updated");
		printf("GCMD:InteractiveConfiguration.Refresh");
		ICUpdate = FALSE;
		ICULast = system_time() + 1000;
	}
}

// Update the persistent memory, but not faster than every 1 second
void PM_Update(uint16 now) {
  static bool PMUpdate = FALSE;
  static uint32 PMULast = system_time();
  if (now=1) PMUpdate=now;  // <---------------------------- yikes
	if (PMUpdate && PMULast < system_time()) { 
		printf("pmem saved");
		PMUpdate = FALSE;
		PMULast = system_time() + 1000;
		pmem_save();
	}
}
#endif
```

Scachi, 2018-08-08:
Same problem happened to another user in the forum, so it is not just me.
After enabling a slot switching loop with two script (multi slots like script) by mistake, with multiple pmem_saves in a script he couldn't start GTuner with the SD-card inserted in the T2 anymore. Formatting the sd-card fixed this issue again.

Scachi, 2018-09-09:
Another one with the problem:
https://www.consoletuner.com/forum/viewtopic.php?f=27&t=9999

J2Kbr, 2018-11-18:
Hi Scachi, could you please confirm if running the above script with the latest firmware/Gtuner available still results on the described issue? I am running the script here for over 15minutes now and Gtuner IV can detect and connect to the device without problems (no error 2). Thanks,.

Scachi, 2018-11-21:
I couldn't reproduce this with the current firmware too. I played for about 30 minutes with a script using this to write all the time to pmem without any problems. SD-card is still working fine.

closed 2018-11-21 by Scachi