J2Kbr/TitanTwo #157: combo and function execution order ?

Scachi, 2018-08-09:
I am not sure if this code ever worked as I thought it would.
Calling a Rapid combo that executes set_val(BUTTON_5,100); wait(100);wait(100);
and using get_val for that button behind the combo call, get_val return always 100 when the user presses the button down, the rapid combo seems to run after the ! get_val code there, pseudo code:

main {
 combo_run(Rapid);
if ( ! get_val(BUTTON_5) ) set_val(BUTTON_21,66);
}

Full post there: 
https://www.consoletuner.com/forum/viewtopic.php?f=26&t=9701&p=68679#p68679

Code used from the Online Resource: "antithesis Antirecoil & Autofire"

Scachi, 2018-08-09:
The execution order of the combos seems to be related to the order the combo 's are defined...

This does not work (not runnable code, just an example):
```
 if (get_val (BUTTON_5)) //antirecoil only active when firing
    {    
          combo_run(RapidFire);
          combo_run(cAntiRecoil);
    }
}

combo cAntiRecoil {
  AntiRecoil(STICK_1_Y,RECOIL_V);
  AntiRecoil(STICK_1_X,RECOIL_H);
}

combo RapidFire
{	
    set_val(BUTTON_5, 100);
    wait(RapidHold);
    set_val(BUTTON_5, 0);
    wait(RapidRest);
}
```

This works:
```
 if (get_val (BUTTON_5)) //antirecoil only active when firing
    {    
		combo_run(RapidFire);
    combo_run(cAntiRecoil);
    }
}

combo RapidFire
{	
    set_val(BUTTON_5, 100);
    wait(RapidHold);
    set_val(BUTTON_5, 0);
    wait(RapidRest);
}

combo cAntiRecoil {
  AntiRecoil(STICK_1_Y,RECOIL_V);
  AntiRecoil(STICK_1_X,RECOIL_H);
}

```

Is this how it should be handled ? Is this documented somewhere ?

Scachi, 2018-08-09:
Hm, putting the rapid combo before the main works too, even without using a combo for antirecoil 

```
 
 //------------------------------------------------------------------------------
// ANTI-RECOIL
fix32 RECOIL_V = 30.0;
fix32 RECOIL_H = 0.0;
fix32 RY;
fix32 RX;

//------------------------------------------------------------------------------
// APEX DEADZONE REMOVER
fix32 StickNoise = 4.32;

//------------------------------------------------------------------------------
// RAPID-FIRE

    // Fine-tune the time between single shots, set to 10 shots / sec by default (100ms)
    uint8 RapidHold = 50; // Hold the button in ms
    uint8 RapidRest = 50; // Release the button in ms
 
 combo RapidFire
{	
    set_val(BUTTON_5, 100);
    wait(RapidHold);
    set_val(BUTTON_5, 0);
    wait(RapidRest);
}
    
 
main {
	
	if (get_val (BUTTON_5)) // only active when firing to allow for microaim adjustments without the input being filtered
	{	
		if(abs(get_actual(STICK_1_X)) < StickNoise) { set_val(STICK_1_X, 0.0); }
		if(abs(get_actual(STICK_1_Y)) < StickNoise) { set_val(STICK_1_Y, 0.0); }
		if(abs(get_actual(STICK_2_X)) < StickNoise) { set_val(STICK_2_X, 0.0); }
		if(abs(get_actual(STICK_2_Y)) < StickNoise) { set_val(STICK_2_Y, 0.0); }
	}
    if (get_val (BUTTON_5)) //antirecoil only active when firing
    {    
    AntiRecoil(STICK_1_Y,RECOIL_V);
    AntiRecoil(STICK_1_X,RECOIL_H);
    combo_run(RapidFire);
    
    }
}

void AntiRecoil (uint8 axis, fix32 recoil)
{

     RY = get_actual(STICK_1_Y);
     RX = get_actual(STICK_1_X);
 
    if (get_val(BUTTON_5) && (sqrt(RX*RX + RY*RY)) <= abs(recoil))
    {
        if(abs(RY) <= abs(recoil))
        {
            set_val(axis,(recoil * (100.0 - abs(get_val(axis)))) / 100.0 + get_val(axis));
        }
    }
}
```

J2Kbr, 2018-10-15:
User defined functions are placed at the end of the bytecode. _main_ block(s) and _combos_ are placed in the bytecode in the same order they are defined in the gpc source code.

closed 2018-10-15 by J2Kbr

Scachi, 2018-10-15:
Is this documented somewhere ? 
I cant find it in the docs.
Adding this to the docs is important as it matters on how the script works.