J2Kbr/TitanTwo #159: Poweroff the device with console - Bluetooth sync issue & user logout

Scachi, 2018-08-25:
I have power off the device with console enabled, there seems to be multiple issues with it..

When I logout of the PS4 to change the account the T2 does shutdown.. can the poweroff with the console setting be improved to not shutdown on user logout ?

When the T2 shuts down on logout of the current user and I do power on the T2 via a single button press on the T2 the Ds4 can't connect anymore via bluetooth..the ds4 just keeps blinking for some time and then shuts down.
When I hold down both buttons of the T2 to go into P mode and leave it again I can connect the ds4 via bluetooth again.
Looks like the BT doesn't wake up correctly after powering down ?

Scachi, 2018-08-26:
This time the bluetooth connection did work fine after pressing one button of the T2 to turn it on again after user logout. So this issue doesn't exist all the time.

J2Kbr, 2018-11-14:
The PS4, by default, turns off the controller when the user logout (i.e. change account). Unfortunately there is no means to detect user logout or console turnoff, at the USB protocol level both cases are exactly the same.  However, if the controller is Bluetooth paired with the Titan Two, you can press the PS button on the controller to restart the Titan Two and resume the normal operations. In this specific case it may take a little bit longer for the controller establish wireless communication with the Titan Two.

closed 2018-11-14 by J2Kbr