J2Kbr/TitanTwo #160: Low-battery Power Warning into Firmware

Scachi, 2018-08-30:
Is it possible to include a battery power warning into the firmware of the T2 ?
So when the battery is going to run out the led or segment display does something or uses the rumbles to indicate that it runs out of power ?
Having to code that into each script is something I don't really like, it is such a basic feature for any battery powered device to run a warning somehow when it runs out of power.

I know that the console's own notification can't be used as the device seems to be connected via usb for the console, so it would be great if the T2 itself can handle it somehow without needing to script that..

Or would that use up to much space of the firmware ?

misterpyrrhuloxia, 2018-10-17:
I second this feature request. This would be very useful and convenient to have.

J2Kbr, 2018-12-03:
With firmware 0.98B-72 the device's display will blink the middle segment to indicate low battery (below 20%) of controller(s).

closed 2018-12-03 by J2Kbr

Scachi, 2018-12-03:
Super cool ! Thank you.