J2Kbr/TitanTwo #170: Feature Request: extend functionality of led_vmset with permanent light mode and intensity setting.

Scachi, 2018-10-27:
As creedkiller suggested in the forum : https://www.consoletuner.com/forum/viewtopic.php?f=26&t=10479&p=73595#p73595

it would be nice to be able to use a single function for controlling the led of the T2 and controller.

At the moment we have to use led_vmset and led_set, depending on if we want blinking or permanent light. Only the later providing intensity.

Personally, I don't use intensity very often, but using a single function for blinking and permanent light might help to save some bytecode in our scripts as we don't have to use different function just for controlling the leds with some if/else statement.

I tried using a long time for the on time of the blinking, zero for off and -1 to fake permanent light but that didn't work reliable in my tests.

J2Kbr, 2018-10-30:
Scachi, you should be able use `led_vmset()` for permanent light as you described, example:

`led_vmset(LED_1, 65000, 0, -1);`

Please let me know what is the issue you are having with these parameters (an example code may help). Thank you.

Scachi, 2018-11-02:
I think I have found the issue in an old script..I was using the command in a combo without a wait(0); behind it so that was the real cause of my trouble.

I have integrated the code like you have written above in several scripts now without any trouble.
Feel free to close this issue. I'll reopen it if I get issues again.

Scachi, 2018-11-02:
I close this for now and will reopen it if I have issues with it again, but I think it was all my fault

Scachi, 2019-02-12:
Well,..problem found again..
according to the documentation this:
```
led_vmset(LED_1, 1000, 0, -1);
``` 
should give permanent color..
The T2 does it..but the controller led shuts off after 1 sec, it is not repeating the cycle.
```
led_vmset(LED_1, 65000, 0, -1);
```
This one only takes longer to shut off^^
It looks like it is only happen when the controller is connected via USB, via Blueooth it is working fine..tested with a ds4 v.(noidea) ..older one.

https://www.consoletuner.com/forum/viewtopic.php?f=26&t=11679&p=79759#p79757

J2Kbr, 2019-03-10:
Cause for this behavior found and fixed on firmware 1.00RC.

closed 2019-03-10 by J2Kbr