J2Kbr/TitanTwo #175: Keyboard detected as Xbox 360 controller

J2Kbr, 2018-11-24:
```
[USB DESCRIPTOR]
12           bLength: 18
01           bDescriptorType: 1 - DEVICE
00 02        bcdUSB: 2.00
00           bDeviceClass: 0 (00)
00           bDeviceSubClass: 0 (00)
00           bDeviceProtocol: 0 (00)
40           bMaxPacketSize0: 64
EB 03        idVendor: 03EB
01 FF        idProduct: FF01
13 01        bcdDevice: 1.13
01           iManufacturer: 1
02           iProduct: 2
00           iSerialNumber: 0
01           bNumConfigurations: 1

09           bLength: 9
02           bDescriptorType: 2 - CONFIGURATION
CE 00        wTotalLength: 206
07           bNumInterfaces: 7
01           bConfigurationValue: 1
00           iConfiguration: 0
A0           bmAttributes: 160 (Bus Powered, Remote Wakeup)
C8           MaxPower: 400mA

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
00             bInterfaceNumber: 0
00             bAlternateSetting: 0
02             bNumEndpoints: 2
FF             bInterfaceClass: 255 (FF)
5D             bInterfaceSubClass: 93 (5D)
01             bInterfaceProtocol: 1 (01)
00             iInterface: 0

11               bLength: 17
21               bDescriptorType: 33 - HID
00 01            bcdHID: 1.00
01               bCountryCode: 1
25               bNumDescriptors: 37
81               bDescriptorType: 129
14 00            wDescriptorLength: 20
00               bDescriptorType: 0
00 00            wDescriptorLength: 0
13               bDescriptorType: 19
02 08            wDescriptorLength: 2050
00               bDescriptorType: 0
00 07            wDescriptorLength: 1792
05               bDescriptorType: 5
81 03            wDescriptorLength: 897
20               bDescriptorType: 32
00 04            wDescriptorLength: 1024
07               bDescriptorType: 7
05 02            wDescriptorLength: 517
03               bDescriptorType: 3
20 00            wDescriptorLength: 32
08               bDescriptorType: 8
09 04            wDescriptorLength: 1033
01               bDescriptorType: 1
00 01            wDescriptorLength: 256
03               bDescriptorType: 3
01 01            wDescriptorLength: 257
00               bDescriptorType: 0
09 21            wDescriptorLength: 8457
11               bDescriptorType: 17
01 00            wDescriptorLength: 1
01               bDescriptorType: 1
22 3B            wDescriptorLength: 15138
00               bDescriptorType: 0
[... 391 lines cut ...]
09 C4            (LOCAL)  USAGE              0x000200C4 Accelerator (DV=Dynamic Value)
09 BA            (LOCAL)  USAGE              0x000200BA Rudder (DV=Dynamic Value)
09 C0            (LOCAL)  USAGE              0x000200C0 Trigger (MC=Momentary Control)
15 00            (GLOBAL) LOGICAL_MINIMUM    0x00 (0)
26 FF00          (GLOBAL) LOGICAL_MAXIMUM    0x00FF (255)
75 08            (GLOBAL) REPORT_SIZE        0x08 (8) Number of bits per field
95 05            (GLOBAL) REPORT_COUNT       0x05 (5) Number of fields
81 02            (MAIN)   INPUT              0x00000002 (5 fields x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap
C0             (MAIN)   END_COLLECTION     Physical
C0           (MAIN)   END_COLLECTION     Application

[INTERFACE 6 HID DESCRIPTOR]
06 FFFF      (GLOBAL) USAGE_PAGE         0xFFFF Vendor-defined
09 01        (LOCAL)  USAGE              0xFFFF0001 
A1 01        (MAIN)   COLLECTION         0x01 Application (Usage=0xFFFF0001: Page=Vendor-defined, Usage=, Type=)
09 02          (LOCAL)  USAGE              0xFFFF0002 
09 03          (LOCAL)  USAGE              0xFFFF0003 
15 00          (GLOBAL) LOGICAL_MINIMUM    0x00 (0)
26 FF00        (GLOBAL) LOGICAL_MAXIMUM    0x00FF (255)
75 08          (GLOBAL) REPORT_SIZE        0x08 (8) Number of bits per field
95 20          (GLOBAL) REPORT_COUNT       0x20 (32) Number of fields
81 02          (MAIN)   INPUT              0x00000002 (32 fields x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap
C0           (MAIN)   END_COLLECTION     Application

```

J2Kbr, 2018-12-03:
Fixed on firmware 0.98B-72

closed 2018-12-03 by J2Kbr

paname75, 2018-12-03:
for this specific keyboard , 
the keyboard when plugged to titan2 even  with last update 98.1.72 doesnt output some keys,  like : 

- spacebar
- - = { }

here is my descriptor fetch 

[USB DESCRIPTOR]
12 01 00 02 00 00 00 40 EB 03 01 FF 11 01 01 02 00 01 09 02 CE 00 07 01 00 A0 C8 09 04 00 00 02 FF 5D 01 00 11 21 00 01 01 25 81 14 00 00 00 00 13 02 08 00 00 07 05 81 03 20 00 04 07 05 02 03 20 00 08 09 04 01 00 01 03 01 01 00 09 21 11 01 00 01 22 3B 00 07 05 83 03 08 00 0A 09 04 02 00 02 03 00 00 00 09 21 11 01 00 01 22 35 00 07 05 85 03 40 00 04 07 05 04 03 40 00 04 09 04 03 00 01 03 00 01 00 09 21 11 01 00 01 22 7B 00 07 05 86 03 40 00 01 09 04 04 00 01 03 00 01 00 09 21 11 01 00 01 22 19 00 07 05 87 03 08 00 10 09 04 05 00 01 03 00 00 00 09 21 11 01 00 01 22 51 00 07 05 88 03 20 00 08 09 04 06 00 01 03 00 00 00 09 21 11 01 00 01 22 17 00 07 05 89 03 40 00 04 
 
[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 10 
10 03 57 00 6F 00 6F 00 74 00 69 00 6E 00 67 00 
 
[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 16 
16 03 57 00 6F 00 6F 00 74 00 69 00 6E 00 67 00 4F 00 6E 00 65 00 
 
[INTERFACE 1 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 01 00 3B 
05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01 75 01 95 00 81 02 81 01 19 00 29 65 15 00 25 65 75 08 95 00 81 00 05 08 19 01 29 05 15 00 25 01 75 01 95 05 91 02 95 03 91 01 C0 
 
[INTERFACE 2 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 02 00 35 
06 37 13 09 01 A1 01 09 02 09 03 15 00 26 FF 00 75 08 95 80 81 02 09 04 09 05 15 00 26 FF 00 75 08 95 80 91 02 09 06 09 07 15 00 26 FF 00 75 08 95 07 B1 02 C0 
 
[INTERFACE 3 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 03 00 7B 
05 01 09 06 A1 01 75 01 95 05 05 08 19 01 29 05 91 02 75 03 95 01 91 03 75 01 95 08 15 00 25 01 05 07 19 E0 29 E7 81 02 75 01 95 2E 15 00 25 01 05 07 19 04 29 31 81 02 75 02 95 01 81 03 75 01 95 69 15 00 25 01 05 07 19 33 29 9B 81 02 75 07 95 01 81 03 75 01 95 08 15 00 25 01 05 07 19 9D 29 A4 81 02 75 01 95 2E 15 00 25 01 05 07 19 B0 29 DD 81 02 75 02 95 01 81 03 C0 
 
[INTERFACE 4 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 04 00 19 
05 0C 09 01 A1 01 75 08 95 01 15 01 26 EA 00 05 0C 19 01 2A EA 00 81 00 C0 
 
[INTERFACE 5 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 05 00 51 
05 01 09 05 A1 01 A1 00 05 09 19 01 29 0C 15 00 25 01 95 0C 75 01 81 02 75 04 95 01 81 03 05 01 09 30 09 31 09 32 09 33 09 34 09 35 16 00 80 26 FF 7F 75 10 95 06 81 02 05 02 09 BB 09 B8 09 C4 09 BA 09 C0 15 00 26 FF 00 75 08 95 05 81 02 C0 C0 
 
[INTERFACE 6 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 06 00 17 
06 FF FF 09 01 A1 01 09 02 09 03 15 00 26 FF 00 75 08 95 20 81 02 C0