J2Kbr/TitanTwo #178: Add support for PS4 HOTAS as an output protocol

NeoTechni, 2018-12-03:
With Ace Combat 7 coming out on PS4 shortly, I'd really like to use one of my existing flight sticks on PS4 since I don't have room or money to get the inferior PS4 models.

CaptMoore, 2018-12-04:
I would like to add my two cents and also ask for flightstick output support for both the Xbox One (emulate the T.Flight HOTAS One), and Xbox 360 (emulate the Saitek AV8R or Ace Combat 6 Hotas Controller.)

If the PS3 had a default flightstick, obviously an output for that console would be great as well, along with the requested ps 4 support. 

The reason why flightstick output is useful (vs just a standard controller emulation) is that some games (such as Ave Combat 6/Hawx) recognize when a flightstick is being used and automatically adjust the control scheme to allow rudder controls to be put on an analog axis.  When using a controller, they put the rudder controls on the digital shoulder button, making precise control difficult.  On hawx (and others I assume) there is no way to manually chose (or override) the control scheme, it is automaticity assigned by the game.

The Xbox one hotas also has more buttons than a standard controller, I believe. 

If it helps, I do have an Xbox one hotas & 2 types of Xbox 360 hotas (ace combat & AV8R) if you need me to volunteer to run sone kind of utility to figure out how they identify themselves as Flightsticks vs standard controllers.

NeoTechni, 2019-01-01:
> The Xbox one hotas also has more buttons than a standard controller, I believe

The Ace Combat one does have a few, but they aren't output to the host. 

The game comes out in 18 days!

swoodbury1963, 2019-06-24:
I sent J2Kbr the PS4 Hotas input descriptors. Hopefully this can be a thing soon. Currently I can't get my T1600M FCS throttle axis or the stick twist axis to work. I can map them and get a good translation in GtunerIV but in Elite Dangerous controller options they aren't working.

swoodbury1963, 2019-07-10:
Checking status.

swoodbury1963, 2019-08-05:
Update please.

swoodbury1963, 2019-08-19:
Thanks for the message reply on the forums!

swoodbury1963, 2019-08-29:
If the other is closed assuming this will be the feature request to check.

thievery80, 2021-04-02:
+1 for output emulation of Thrustmaster T-Flight HOTAS ONE/4 on XBOX/PS

-> If any help with HID´s for the XBOX model is needed, happy to help!

swoodbury1963, 2021-04-03:
enjoy my friend's early bird preview...

https://drive.google.com/file/d/1oTOaGaTAoy_oButZSG8lATBhfREqNlsw/view?usp=sharing

CaptMoore, 2021-04-03:
Is the preview for Xbox series/one/360 or PlayStation?

swoodbury1963, 2021-04-08:
No, it's not. It's the thrustmaster Thrustmaster Tm1600 HOTAS Throttle