J2Kbr/TitanTwo #179: OTG Adapter in Prog port won't authenticate PS4

francis-specialeffect, 2018-12-07:
The TITAN Two won't authenticate on a PS4 with an official PS4 controller connected to the Prog port via the OTG Adapter. Is this intended? I had thought it did previously.

J2Kbr, 2018-12-12:
This is correct. The PROG port with OTG adapter can be used to connect HID devices (such as joysticks, keyboard, mouse, wheels etc), but not controller from consoles. Internally the Titan Two have a micro-controller and an high-speed USB controller. The inputs A, B and the OUTPUT are handled by the USB controller, and the PROG is directly connected to the micro-controller.

When using a controller compatible with the connected console, the Titan Two creates a direct connection between the controller and the console. This allows the Titan Two to support any current and future console specific features (such as headsets, chatpads, etc) without any internal implementation.

However, this direct connection (called full passthru mode) is only possible within the USB ports on the USB controller. That is why the PROG only supports HID devices, as it can't be used for full passthru mode.

If the Titan Two is equipped with Bluetooth module, you can have the PS4 controller connected to the Titan Two wirelessly for authentication purposes.

closed 2018-12-12 by J2Kbr

oneswitch, 2020-01-28:
Fixed. Thank you!