J2Kbr/TitanTwo #180: Rapid Connect and Disconnect PS4/DS4

Hyphen-x, 2018-12-18:
I have **new T2 on the way** I replaced the 1st one and **I want to setup it up carefully.** 
to be able to troubleshoot back any steps **GTuner version 10216 is one I have that works** with my version of windows **do you recommend in my setup trying the latest version and is there anything I need to be cautious about because of these issues I had with the T2 I had.** 

With the first T2 I started having problems with a cable issue. 

I was using the cable that came with the T2 for output to PS4 and the long cable from the T2 kit USB A to controller. This setup worked great and I was able to use anything online resources had to offer. 

**I did a cable change** and put the long cable in output to the PS4 and switch a couple of others. The in game menu select lights started not responding when trying to use ghosts pack in game and then the T2 stopped responding I did get some help that correct the menu activation sequence I was using and that worked but script started acting up and responding glitchy. 

To address this issue **the software was updated on GTuner IV** that I had at the time and I forced an **update on the T2** it was also connected. I put T2 in P mode and updated in GTuner IV. 

After working on these issues not being connected to the PS4 I turned it on for testing and that is when the **rapid cycling with the notifications of disconnect without USB icon and the reconnect with USB icon started** it was sporadic but after trying to correct it with the data link cable it slowed down but still interrupted the input signals.to the PS4. It was at every 20 secs last time I tried. 

**I had GTuner version:10216 maybe** but this one does work and the latest one did not work and caused a problem with drag and drop 10216 fixed that. I have **Windows 10 1607**

Hyphen-x, 2018-12-19:
I think the **problem on the GTuner IV drag and drop for my self** is to rule out that **Windows 10 v.1607** version (the last version before Microsoft product validation tool was forced on users) I have an old copy of MS office it makes it shuts down. So That may be the case with others too (I wanted a hybrid of Windows 10 and 7 so I had to freeze it there.  I put my **WiFi connection on metered and this stops Microsoft updates**. 
So I do not know if i just need to keep running the older version you gave me of GT which works fine if **that is okay to do everything**. or I may clone my drive onto my D. drive and create a dual boot.. so I can use my office 2010 on one and **update the original to latest windows 10 version if it mattered** on having the latest version of GTuner IV. **What is anyone thoughts** on this because **only the developer knows if that matters** I don't but **I suspect** it does because of the hooks (special rewrites) you have to have to use new features in windows newer versions T2 is not backward compatible.

Hyphen-x, 2018-12-19:
On the disconnect and reconnect issue my new **T2 arrived last night** and I am nervous about hooking it up.. **I can confirm** the **Drag ad Drop problem still occurs with the new version of GT** and that t**he older version 10216 still allows drag and drop**. I have not connected it to the PS4 yet. I will use the **cable that came with it for output/PS4**, usb is enable in PS4, The long cable that came in the cable kit **Port A/Controller**/ and a **long micro usb cable I have to go from prog/computer**. I have information that says it does not matter when using one controller whether i start with it in port a or some instructions say port b. **I am using port A because that is how i did it the first time and it worked** fine till i switched cables and updated. Also, **another factor** was when I started trying to do **configuration changes in GT while live in PS4** game at the same time I updated to try to learn the GT abilities. I do not place a lot of weight myself on any of these factors i suspect and hope the firmware rollback? will probably fix it but if it does not I may cry. :-) will put it together later (Christmas duties call)

Scachi, 2018-12-19:
The old GT IV version I gave you should not be used to compile scripts. Please use the actual version for that.  The old version doesn't support newer features and isn't 100% compatible to the current firmware.

The old version should work fine for the drag and drop only available scripts.
As for those drag and drop only scripts there is nothing to compile, they are already compiled.

Hyphen-x, 2018-12-19:
Yes but when I add them using the old GT4 and then bring it back in it does lose the configuration ability... ...  **I am hoping someone can confirm that it is not my version of windows 10** causing the problem :-(  (drag and drop) because it would take me a half a day to put two versions of my system files in a dual boot config to use the latest version... not something i planned on having to do. Does all the new stuff work with windows 7 or how is that addressed for windows 7 users does anyone know.

Scachi, 2018-12-19:
Total dumb question.. did you do some changes to your electricity network in your house/apartment ?
Adding a power hungry device like a heater or plugged in your power cable of your ps4 into another socket than before ? 
Just asking as I had trouble when using my 3D printer from the same power socket as my ps4 or hifi. Getting noise and disconnects. Depending on the cables it can be enough to place an usb cable to near to a power supply cable of a high power device to cause problems like flicker on monitors and stuff like that.
I had a lot of "fun" with something like that at work.

Do you have the chance to test the T2 on your ps4 in another room using another power socket ?
Next level would be for me if I had the problems to test it with another ps4 system when visiting a friend, at best with the friends cables to rule everything else out. If this works without issues on your friends system it may help  to get an idea what to look for as a possible cause.

Hyphen-x, 2018-12-20:
@Scachi

**Go to the bottom to see latest outcome/status/progress. I think i found the disconnect/reconnect current cause. And the drag and drop problem for me may tie into this other problem too. so I put my testing info up to help.** 

I received my new T2 replacing the old one and I set it up for testing during its setup.

Re: T2 official known issues thread

Post by Hyphen- » Wed Dec 19, 2018 2:22 pm
I have connected the new T2 to the PS4 .. it is sleeping quietly in the corner and no disconnects from the PS4 going past 1 hour...
I am about to connect it to my computer to view it in GT4 newest version first. will update afterwards...

Post by Hyphen- » Wed Dec 19, 2018 2:44 pm
I connected the T2 setup while idling on PS4 to my computer. The newest release of GT4 would not drag and drop into the T2 memory slots on my Windows 10 v.1607. And all items I had put on with an earlier version lost their configure features and the gear icon disappeared. There still is no disconnect/reconnect issues.

Post by Hyphen- » Wed Dec 19, 2018 3:12 pm
While setting my working directory still in the latest GT4 I decide to see what would happen if I unplugged the usb cable on my computer connected to the prog port.

The disconnection Triggered here:

When I did unplug the usb cable from my pc with my DS4 in port B on T2 it disconnected from the PS4 showing the cycling notifications and when I plugged it back in it stopped. And I have repeated it and it occurred again when i unplug the T2 from my computer. But it has not reoccurred while it is plugged in to the computer while GT4 is actively connected to T2 and if I do not unplug it.

My next test will be to turn off GT4 then unplug it but I will enjoy this moment for awhile...

I want to test the older version of GT4 too although I have been advised it will not work properly with the new T2 firmware but not officially from @J2Kbr.

I do not want to unplug from the computer now lest i trigger a more permanent problem. It does not cycle at this point but it may start once i unplug. it. :roll:

Set up: micro usb cable that came withT2 PS4/output, long cable in cable kit from port B to DS4 latest new, A long micro usb cable from prog/computer (mine).

Edit: Also I have the T2 connected to a recommended power supply.

Post by Hyphen- » Wed Dec 19, 2018 3:25 pm
I turned off GT4 (closed it) and pulled the usb cable out on my computer... and it did not disconnect the controller in port B from the PS4. So that was interesting.

While the T2 was connected to GT4 latest version and I unplug the T2 from the computer while it is connected it caused the Controller disconnect and reconnect problem. I did not leave it on to see if it would start cycling because I don't want to order another one. Okay what does this mean and can i get a vesion of GT4 that I can use in my windows 10 version I froze it there for a lot of reasons.

Post by bonefisher » Wed Dec 19, 2018 3:46 pm
Check in the device configuration tab on bottom right of GtunerIV while it is plugged in look at power source: to see what is giving the power at this time. Right now mine says DC power jack since my T2 is drawing power from that. If yours says prog. port this is why it disconnects at the time you unplug.

Post by Hyphen- » Wed Dec 19, 2018 4:07 pm
After disconnecting the T2 from the PC with GT4 newest version program closed. And the T2 active and connected to PS4 home page since the last post 35 minutes ago it has not disconnected the controller from the PS4. I am running out of time to test this but that should give some direction.

My main question is still is there a version of GT4 that will fix the drag and drop problem with my version of Windows 1607 or older versions or do what versions are you compatible with because to rule that out I would have to reconfigure my hard drives to use the latest Windows 10 which isn't that great.Thanks I am still having fun! :whistling:

Post by Hyphen- » Wed Dec 19, 2018 4:17 pm
@J2Kbr Or you can develop a way to run it in compatibility mode for older versions I have seen that before in script programs (Java) becoming incompatible often as windows changes so rapidly... but I will need a work around to use the latest version of GT4 if you have ideas. Thanks again

bonefisher wrote:
**Check in the device configuration tab on bottom right of GtunerIV** while it is plugged in look at power source: to see what is giving the power at this time. Right now mine says DC power jack since my T2 is drawing power from that. If yours says prog. port this is why it disconnects at the time you unplug.

Hey @bonefisher, I have it hooked up to the power supply and it does show up that way in GT4 the problem is caused by the pc maybe as an unwanted power source disrupting the input on contact some how. Check my next test.

update:

I did some updating to my copy of windows

Windows 10
2018-05 Update for Windows 10 Version 1607 for x64-based Systems (KB4132216)
2018-07 Cumulative Update for Windows 10 Version 1607 for x64-based Systems (KB4346877)
Fixed some framework problems/denials on com ports but wanted to do this before my next test.

Same setup after update the disconnection/connection problem changed and became a bit more focused

I **now can cause the problem to happen on demand** 1, by slowly putting in the usb in its slot to make the connection on the pc from the prog port. As soon as the usb makes contact the both green lights go red and the notification pops up. It happens on  [... cut ...]
the drag and drop still does not work and I once again cannot access the gamepacks that are drag and drop. :roll:

**But the cycling is not looping continuous like it did before.** _I have not loaded any scripts and tried to configure them in game yet and I have a feeling that is when problems would happen possibly._ **I can use the configuration program on the old version of GT4 but when I load the T2 into the newest one the configuration capabilities disappear there is no gear icon**. :unsure:

Hyphen-x, 2018-12-20:
Re: T2 offical known issues thread

Post by Hyphen- » Wed Dec 19, 2018 11:26 pm
Current update:

**I used the older version of GT4** that allows me to drag and drop gamepacks into T2 and
**The setup** short cord the came with T2 output/PS4, long cable from kit Port B/DS4, long cable (mine) T2 prog/PC, and power supply.
**Loaded infinite warfare pack** and configured it. The T2 responded well. Everything worked fine in game and **I was able to configure the game pack in real time** using the older version of GT4 and t**here were no disconnect/reconnect issues**. I left everything connected throughout.

So I can say that **at this point the T2 is working normally** _except_ for **I cannot use the latest version of GT4 the drag and drop does not work yet** at this time.

J2Kbr, 2018-12-20:
Issue fixed with Gtuner IV 1.00RC-86 (released today). The problem was with the Qt framework 5.12.0, which has a bug with drag-and-drop:

https://forum.qt.io/topic/97649/bug-in-qt-5-12-drag-n-drop

The Qt framework was downgraded to 5.11.3.

closed 2018-12-20 by J2Kbr