J2Kbr/TitanTwo #184: (Feature Request) Wireless BT connection support for the GameSir G5 controller

DaFlorian, 2018-12-25:
Hello,

I bougth the GameSir G5 controller because it has a lot of buttons which can be possibly configured.
Unfortunatly no wireless connection is possible to the Titan Two device (with expansion kit).
The devices won't pair. USB connection works fine.

Would it be possible to add wireless support for the controller and what infos do you need to accomplish this task.

Complete setup:
Device: Titan Two device + wireless expansion kit (latest firmware)
Console: Xbox One X
Controller: GameSir G5

Thanks in advance

DaFlorian, 2019-01-12:
As I wrote in the forum...
USB-Connection to the controller is possible.
Here are the Descriptors captured by Gtuner IV for the GameSir G5 controller:

[USB DESCRIPTOR]
12 01 00 02 FF FF FF 08 5E 04 8E 02 05 03 01 02 03 01 09 02 99 00 04 01 00 A0 FA 09 04 00 00 02 FF 5D 01 00 11 21 00 01 01 25 81 14 00 00 00 00 13 06 08 00 00 07 05 81 03 20 00 05 07 05 06 03 20 00 08 09 04 01 00 04 FF 5D 03 00 1B 21 00 01 01 01 83 40 01 04 20 16 82 00 00 00 00 00 00 16 05 00 00 00 00 00 00 07 05 83 03 20 00 02 07 05 04 03 20 00 04 07 05 82 03 20 00 40 07 05 05 03 20 00 10 09 04 02 00 01 FF 5D 02 00 09 21 00 01 01 22 87 07 00 07 05 87 03 20 00 10 09 04 03 00 00 FF FD 13 04 06 41 00 01 01 03

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 0E
0E 03 58 00 69 00 61 00 6F 00 6A 00 69 00

[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 16
16 03 47 00 61 00 6D 00 65 00 73 00 69 00 72 00 2D 00 47 00 35 00

[STRING 3 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 03 04 09 00 16
16 03 31 00 32 00 33 00 34 00 35 00 36 00 37 00 38 00 39 00 30 00

J2Kbr, 2019-01-15:
Thank you for sending the descriptors (above from XB360 compatibility mode). I also see your sent the HID descriptors (which are the ones we are interested) on the forums, here is for future reference:

```
[USB DESCRIPTOR]
12           bLength: 18
01           bDescriptorType: 1 - DEVICE
10 01        bcdUSB: 1.10
00           bDeviceClass: 0 (00)
00           bDeviceSubClass: 0 (00)
00           bDeviceProtocol: 0 (00)
08           bMaxPacketSize0: 8
AC 05        idVendor: 05AC
7A 05        idProduct: 057A
05 03        bcdDevice: 3.05
06           iManufacturer: 6
07           iProduct: 7
00           iSerialNumber: 0
01           bNumConfigurations: 1

09           bLength: 9
02           bDescriptorType: 2 - CONFIGURATION
42 00        wTotalLength: 66
02           bNumInterfaces: 2
01           bConfigurationValue: 1
00           iConfiguration: 0
A0           bmAttributes: 160 (Bus Powered, Remote Wakeup)
FA           MaxPower: 500mA

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
00             bInterfaceNumber: 0
00             bAlternateSetting: 0
02             bNumEndpoints: 2
03             bInterfaceClass: 3 (03)
00             bInterfaceSubClass: 0 (00)
00             bInterfaceProtocol: 0 (00)
00             iInterface: 0

09               bLength: 9
21               bDescriptorType: 33 - HID
11 01            bcdHID: 1.11
00               bCountryCode: 0
01               bNumDescriptors: 1
22               bDescriptorType: 34
A1 00            wDescriptorLength: 161

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
02               bEndpointAddress: 2 OUT
03               bmAttributes: 3 (Interrupt, None, Data)
40 00            wMaxPacketSize: 64
10               bInterval: 16

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
82               bEndpointAddress: 2 IN
03               bmAttributes: 3 (Interrupt, None, Data)
10 00            wMaxPacketSize: 16
10               bInterval: 16

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
01             bInterfaceNumber: 1
00             bAlternateSetting: 0
01             bNumEndpoints: 1
03             bInterfaceClass: 3 (03)
01             bInterfaceSubClass: 1 (01)
02             bInterfaceProtocol: 2 (02)
00             iInterface: 0

[... 110 lines cut ...]
A1 00          (MAIN)   COLLECTION         0x00 Physical (Usage=0x00010001: Page=Generic Desktop Page, Usage=Pointer, Type=CP)
85 01            (GLOBAL) REPORT_ID          0x01 (1) 
09 30            (LOCAL)  USAGE              0x00010030 X (DV=Dynamic Value) 
09 31            (LOCAL)  USAGE              0x00010031 Y (DV=Dynamic Value) 
09 38            (LOCAL)  USAGE              0x00010038 Wheel (DV=Dynamic Value) 
75 08            (GLOBAL) REPORT_SIZE        0x08 (8) Number of bits per field 
95 03            (GLOBAL) REPORT_COUNT       0x03 (3) Number of fields 
15 81            (GLOBAL) LOGICAL_MINIMUM    0x81 (-127) 
25 7F            (GLOBAL) LOGICAL_MAXIMUM    0x7F (127) 
81 06            (MAIN)   INPUT              0x00000006 (3 fields x 8 bits) 0=Data 1=Variable 1=Relative 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
15 00            (GLOBAL) LOGICAL_MINIMUM    0x00 (0) 
25 01            (GLOBAL) LOGICAL_MAXIMUM    0x01 (1) 
05 09            (GLOBAL) USAGE_PAGE         0x0009 Button Page 
19 01            (LOCAL)  USAGE_MINIMUM      0x00090001 Button 1 Primary/trigger (MULTI=Selector, On/Off, Momentary, or One Shot) 
29 03            (LOCAL)  USAGE_MAXIMUM      0x00090003 Button 3 Tertiary (MULTI=Selector, On/Off, Momentary, or One Shot) 
75 01            (GLOBAL) REPORT_SIZE        0x01 (1) Number of bits per field 
95 03            (GLOBAL) REPORT_COUNT       0x03 (3) Number of fields
81 02            (MAIN)   INPUT              0x00000002 (3 fields x 1 bit) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
75 05            (GLOBAL) REPORT_SIZE        0x05 (5) Number of bits per field 
95 01            (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields 
81 03            (MAIN)   INPUT              0x00000003 (1 field x 5 bits) 1=Constant 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
C0             (MAIN)   END_COLLECTION     Physical
C0           (MAIN)   END_COLLECTION     Application

```

J2Kbr, 2019-04-06:
Added HID Configuration file for GameSir G5 (05AC:057A) on the next Gtuner IV update.
To use be sure to:
`Device Configuration -> SD-Card -> Configure`
After the update release.

closed 2019-04-06 by J2Kbr

sam0383, 2022-01-11:
how to start the joystick?
after pressing the reset button, it stopped turning on (Does not respond to pressing the power button, and the indexer does not light up). You need a full firmware of the joystick using OTG, since it does not connect to android devices via bluetooth. when connected to the phone with a cable, he does not see it. When connected to a Windows computer, it appears as a device but does not work.

requires primary firmware for the joystick