J2Kbr/TitanTwo #186: Xbox one controller randomly disconnects from Xbox wireless adapter

syntactic-salt, 2018-12-30:
The Xbox One controller disconnects randomly from the Xbox wireless adapter. Sometimes it stays connected for hours and sometimes it disconnects several times in a matter of a couple minutes. Everything works fine if you take the wireless adapter out of the equation and use a USB cable.

syntactic-salt, 2019-03-28:
@J2Kbr what's the status of this issue?

J2Kbr, 2019-04-06:
This next week we will be releasing a new firmware update with several improvements. When available, please update your Titan Two and check if any improvement can be noticed on the connection stability using the Xbox Wireless Adapter.

_Side note:_ in my testing with Xbox Wireless Adapter the connection is rock solid.

closed 2019-04-06 by J2Kbr

syntactic-salt, 2019-04-07:
@J2Kbr I'll give it a try when the update comes out.

Just to be clear, I've followed all the instructions and there's obviously an issue. Just because you can't reproduce it doesn't mean it doesn't exist. Others in the forums have reported it. Why would you close the issue and not work with me to figure out the problem with the TitanTwo? If it's not fixed after the updates, I'm just going to open another issue, so I'm not sure what the point is other than to try to get me to just go away.

For what it's worth, I think the issue is the wifi switching. Is the TitanTwo handling the wifi switching like it should to support the Xbox Wireless Adapter?