J2Kbr/TitanTwo #188: 3DRudder + Xbox Controller = broken triggers

AeliusSaionji, 2019-01-11:
Hello!

With the 3DRudder plugged into InputA and the Xbox One Elite controller plugged into InputB, everything works as expected- except the triggers. According to Gtuner, both triggers are locked to the value `50`. Perhaps the TitanTwo is misreading something from 3DRudder, and blocking any trigger input from the controller?

All functions of the 3DRudder are detected and working with TitanTwo, so it's strange that the xbox triggers are not working.

J2Kbr, 2019-04-06:
Please capture the USB descriptors of your 3DRudder, so I can check the compatibility with the Titan Two.

On Gtuner IV, with the Titan Two connected to computer, go to top menu Tools -> "Fetch Descriptors ...". Now plug the 3DRudder, to any port on the Titan Two. The descriptors will show in the window, click on the button "Copy Descriptors" and post here. Thank you.

AeliusSaionji, 2019-04-06:
3DRudder in Joystick mode (default)
```ini
[USB DESCRIPTOR]
12 01 00 02 00 00 00 08 FA 2D 01 00 19 14 01 02 00 01 09 02 29 00 01 01 00 C0 32 09 04 00 00 02 03 00 00 00 09 21 11 01 00 01 22 7C 00 07 05 81 03 40 00 01 07 05 01 03 40 00 01 

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 1C 
1C 03 33 00 44 00 52 00 75 00 64 00 64 00 65 00 72 00 20 00 49 00 6E 00 63 00 2E 00 

[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 12 
12 03 33 00 44 00 52 00 55 00 44 00 44 00 45 00 52 00 

[INTERFACE 0 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 00 00 7C 
06 00 FF 09 01 A1 01 85 01 15 00 25 FF 75 08 95 3F 19 01 29 3F 81 02 19 01 29 40 95 40 91 00 C0 05 01 09 05 A1 01 85 03 05 01 25 FF 45 FF 09 00 75 08 95 01 81 02 15 00 25 01 35 00 45 01 75 01 95 10 05 09 19 01 29 10 81 02 05 01 09 30 09 31 09 32 09 33 09 34 09 35 09 36 09 36 15 00 26 FF FF 35 00 46 FF FF 75 10 95 08 81 02 19 01 29 40 15 00 26 FF 00 75 08 95 40 91 00 C0 
```

AeliusSaionji, 2019-04-06:
Maybe I missed this before.

It seems that it's just reading this button press information from 3DRudder. This happens without the xbox controller plugged in. It unfortunately also overrides the xbox controller.

Here's a video going through all of 3DRudder's axis.

https://aelius.keybase.pub/3drudder-gtuner.mp4

AeliusSaionji, 2019-04-06:
Now that I know how to use GtunerIV better, I figured out I can work around this problem by making use of an input translator and the port remapper mask like so:

[image]

Only problem is, now I need to learn GPC coding :wink: to also disable the Elite's paddles, since I don't think I can have two input translators running at once.

3DRudder also has a "HOTAS" mode, which has a different USB descriptor:

```ini
[USB DESCRIPTOR]
12 01 00 02 00 00 00 08 FA 2D 01 00 19 14 01 02 00 01 09 02 29 00 01 01 00 C0 32 09 04 00 00 02 03 00 00 00 09 21 11 01 00 01 22 A1 00 07 05 81 03 40 00 01 07 05 01 03 40 00 01 

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 1C 
1C 03 33 00 44 00 52 00 75 00 64 00 64 00 65 00 72 00 20 00 49 00 6E 00 63 00 2E 00 

[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 12 
12 03 33 00 44 00 52 00 55 00 44 00 44 00 45 00 52 00 

[INTERFACE 0 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 00 00 A1 
06 00 FF 09 01 A1 01 85 01 15 00 25 FF 75 08 95 3F 19 01 29 3F 81 02 19 01 29 40 95 40 91 00 C0 05 01 09 04 A1 01 85 03 05 01 25 FF 45 FF 09 00 75 08 95 01 81 02 09 01 A1 00 15 00 26 FF FF 09 30 09 31 09 32 09 33 09 34 09 35 15 00 26 FF FF 75 10 95 06 81 02 09 36 15 00 26 FF 00 75 08 95 04 81 02 05 09 19 01 29 22 15 00 25 01 95 22 75 01 81 02 75 02 95 01 81 01 05 01 09 39 15 01 25 08 35 00 46 3B 01 66 14 00 75 04 95 01 81 42 05 05 09 24 09 26 15 00 25 0F 75 04 95 02 81 02 C0 C0 
```

and in HOTAS mode, ACCEL_1_X is locked to -100. In this mode, button21 is not highlighted.

[image]

J2Kbr, 2019-04-07:
Thank you for the provided information. In any case next Gtuner IV release adds HID configuration file for 3D Rudder (2DFA:0001).

Reconfigure your SD-Card (Device Configuration -> SD-Card -> Configure) after the updating your Gtuner IV.

closed 2019-04-07 by J2Kbr

AeliusSaionji, 2019-04-08:
Looks like this bug (or is it the 3DRudder spitting out bad data?) is still present unfortunately. The triggers are no longer locked to 50, but they are still highlighted in Gtuner- and they _are_  detected as pressed by the WiiU.

[image]

Still new at GPC, but I managed to use it to debug.

```C
fix32 trig;
main {
    trig = get_actual(BUTTON_5);
    printf("%.10f", trig);
    if (check_active(BUTTON_5, 1)){
        printf("active");
    }
    else {
        printf("inactive");
    }
}
```

output:
```
GVM Output: active
GVM Output: 0.0015258789
```

Original issue is still present: triggers from xbox controller are ignored because of 3DRudder. Also, since there is still some value > 0, any GPC function relating to checking if the triggers are "active" will always return true. For the moment I can work around this by simply unbinding the problem buttons on InputA, no rush :)

Maybe unrelated, but the KMG capture won't work for 3DRudder either.

[image]

Thank you for looking into this!

J2Kbr, 2019-04-08:
Thank you for reporting the results with the 3D Rudder HID configuration file. Definitively we are making progress, now just needing fix the remaining 0.0015258789 residual value by offsetting it. I will try get this done for the next Gtuner IV release.

AeliusSaionji, 2021-01-10:
Hello!
Sorry to bother you, but this is coming up in my setup again.
Did it end up being a problem to squelch this?

Using a higher precision I was able to see that the exact offset reaches out a bit further than previously reported.

`GVM Output: 0.00152587890625000000000000000000000000000000000000`

for both button's 5 and 8.

Thanks!