J2Kbr/TitanTwo #189: KMG Capture: Upgrade Request for Key > Variable Analogue Output

oneswitch, 2019-01-16:
Last night, I had speech and EyeGaze* working well with the Titan Two on a PS4 using KMG Capture. Really nice. Brilliant what you can do with the T2.

I can map WASD to 100% left-stick movement for example. I've mapped all the basic controls in this way.

I can't map key-strokes to analogue to act as anything less than 100%. 

If I could it would be immensely useful. For example, by saying "walk slowly" into GavPie speech to key-strokes software, have that translate to outputting "SHIFT+W" (toggled on/off). Have GTuner's KMG Capture detect and translate that to -25% LY Stick 2 movement for example.

Speech using GavPie - and EyeGaze using a Tobii 4C and Project Iris: http://iris.xcessity.at - both emulating key-strokes.

oneswitch, 2019-01-24:
Actually, modifiers not necessary I suppose. This is something I'd love to be able to replicate in the Input Translator:

Left-Stick Slow ←→↑↓ = **E R T Y**   (e.g. pressing T = left-stick up 55%)
L-Stick Slowest ←→↑↓ = **V B N M** 

Right-Stick Slow ←→↑↓ = **; # / ’**
R-Stick Slowest ←→↑↓ = **Home End PgUp PgDn**

Left Trigger/L2 slow to slowest = **O [ .** 
Right Trigger/R2 slow to slowest = P ] ,

J2Kbr, 2019-04-06:
Copy of the email I sent you regarding this request.

> I am aware this feature is long waited by you. I have already put some time on your request, specially to try find a way to implement on top of what we have already working, keeping compatibility with the many published Input Translators we have already available on Gtuner IV. 
> 
> Unfortunately I didn’t find an easy solution that could be applied only to the Input Translator layer. If I remember correctly, we did something related to this using scripts right? How this is working? And what are the limitations you may have using this approach?

closed 2019-04-06 by J2Kbr

oneswitch, 2019-04-08:
Hi Jefferson. Thanks for trying.

We're still working on it, with very tight space constraints. Think we'll be able to merge an input translator KMG file alongside some scripts to convert key-presses to different analogue values in a useful way. Fingers crossed.

Limitations of having to code it in script is that we're nearly completely out of memory, and other valuable features will have to be excluded.

Cheers. Barrie