J2Kbr/TitanTwo #191: Xbox Adaptive Controller Bluetooth Support

francis-specialeffect, 2019-01-21:
Currently unable to connect the [Xbox Adaptive Controller](https://www.xbox.com/en-US/xbox-one/accessories/controllers/xbox-adaptive-controller) to the Titan Two via Bluetooth, tested on multiple controllers and systems. Would it be possible to add Bluetooth support similar to a standard Xbox One Controller?

J2Kbr, 2019-03-10:
Firmware 1.00RC will load the Xbox One driver when found the Xbox One Adaptive controller over Bluetooth. This should be enough to get the controller to work wirelessly (assuming the Bluetooth protocol is the as the S controller).

closed 2019-03-10 by J2Kbr

francis-specialeffect, 2019-03-11:
In that case I think it must use a different Bluetooth protocol. I have just tested pairing with an Adaptive Controller and an S Controller and only the S controller will pair with the Titan Two.