J2Kbr/TitanTwo #193: High CPU usage > 100 % on keybaord functions

Scachi, 2019-01-24:
Mouse movement and key_status check does use to much cpu power.
I don't understand how/why mouse events to increase the key_status cpu usage somehow ?

Forum post: https://www.consoletuner.com/forum/viewtopic.php?f=27&t=11560

Some testings I have done: https://www.consoletuner.com/forum/viewtopic.php?f=27&t=11560&p=78665#p78652

Scachi, 2019-02-07:
Here is another script with a problem that looks similar.
The cpu usage gets higher (75% and more) as soon as a second for loop is run and the mouse moves.
To see the difference run the attached script with the function fNullOutput as it is attached.
After that comment the active code block in the function fNullOutput  and uncomment the lower block in the function fNullOutput   

[**Fenria.zip**](https://github.com/J2Kbr/TitanTwo/files/2842825/Fenria.zip)

Is this just a cpu power "normal limitation" of the T2 that does that on two for loops and mouse movement or with a ds4 controller connected. 
When doing the for loop "manually"..writing all the if statements myself instead of using the for loop it has no cpu usage.
Without a controler connected or without moving the mouse the cpu usage looks fine  no matter which fNullOutput version I use.

J2Kbr, 2019-03-10:
I tested and optimized the example code. The high CPU usage was mainly cause for multiple calls of key_check() and due also a bug in the key_event_active() function. Going post the optimized code in the forums.

closed 2019-03-10 by J2Kbr