J2Kbr/TitanTwo #195: [Request] Xinput with USB Multi Interface HID

Dext1, 2019-02-25:
Any chance to have an option to output gamepad functions in Xinput instead of DirectInput? A lot more titles on PC support Xinput and it would be nice to use mouse/keyboard&gamepad inputs simultaneously in titles that support it.

J2Kbr, 2019-03-10:
Xinput are for Xbox 360 and Xbox One controllers. The Titan Two can already output emulating any of theses two controllers. However, your goal is output as K&M at the same time. On the Titan Two the K, M and Gamepad output interfaces are handled by the HID emulator.

Unfortunately the Xbox controller protocols are not HID fully compliant, which would require re-implementation of theses protocol inside of the HID emulator. Considering we have only 19% of the flash-memory on the Titan Two free for new features (firmware), is better avoid re-implementation of existing codes.

The good news is, there are many software solutions that can convert DirectInput to Xinput, here is a free/open source example:

https://github.com/csutorasa/XOutput

closed 2019-03-10 by J2Kbr

sillyasskid, 2019-03-11:
You can use [x360ce](https://www.x360ce.com/), to convert the Titan Two HID Direct Input Gamepad to a Xbox360 controller.

I have had success using this on the PC version of GTA V.