J2Kbr/TitanTwo #197: led_vmset issue - doesn't set the controller color on the first try 100% of the time

Scachi, 2019-03-14:
Using the code below and an older ds4 controller connected via bluetooth or usb, doesn't matter, running via test&debug or from internal slot, doesn't matter, the controller led doesn't always match the color of the T2 all the time on the first try. It fails like 2 out of 4 time for me.
The T2 displays the color correctly, the controller showing deep blue (ps4 is not running).

Code:
```
init {
  led_vmset(LED_3, 65000, 0, -1); // green .. 3 out of 4 tries
}
```

J2Kbr, 2019-04-06:
I was not able to reproduce. Tested several times with both DS4 models, wired and wirelessly. Every single attempt the lightbar of the controller changed to green, as expected.

However, the firmware I am running has several changes/updates not yet available on the currently released version. These changes may have a positive effect on the reported behavior.

I am closing this issue for now, please reopen if the problem persists with the next firmware release.

Thank you.

closed 2019-04-06 by J2Kbr

Scachi, 2019-04-10:
The latest firmware update has fixed the issue. It is working fine now in all cases.