J2Kbr/TitanTwo #199: PS4 output mode no longer works with VitaTV

NeoTechni, 2019-03-20:
I could have sworn it worked before. But now it only works in PS3 mode, which means I can't use any touchpads.

J2Kbr, 2019-04-06:
The PS4 output protocol never worked with Vita TV, as the console forces the controller to operate over Bluetooth despite the USB connection. However, with the support for Bluetooth connection with PS4 console (recently released), we were able to extend this feature also to the Vita TV. Starting from next firmware update, you can connect with the Vita TV using the `Bluetooth PlayStation 4` output protocol. Please note you will need first pair the Titan Two with the Vita TV, as described here:

https://www.consoletuner.com/wiki/index.php?id=t2:usage_guides:systems:playstationtv#wireless_connection_as_dualshock_4

closed 2019-04-06 by J2Kbr

NeoTechni, 2019-04-06:
Thank you. I discovered the VitaTV's tomfoolery shortly after posting that.
I am glad you were able to fix another of the VitaTV's flaws