J2Kbr/TitanTwo #205: led_set intensity

AJSinc, 2019-05-08:
Code such as: 
`
 init {
	 led_set(LED_1, 0.0, 0); // causes Xbox LED to blink
	 //led_set(LED_1, 0.001, 0); // causes Xbox LED to be off, and not blink
 }
 
 main {

 }
`

causes the Xbox LED to blink. Based on the documentation of the led_set function, the intensity parameter: 
"intensity: The LED intensity level in percentage, from 0.00 to 100.00 (%). Pass a negative value to use the “Default LED Intensity” configured by the user in the device settings."

should not cause the LED to blink. Setting the intensity parameter to 0.001, however, causes the LED to properly shut turn off.

J2Kbr, 2019-06-16:
Possible some other Player LED (P2, P3 or P4) has the intensity and blinking parameter set. To ensure the Xbox LED to turn off set all LEDs to 0, as the following:
```
init {
    led_set(LED_1, 0.0, 0);
    led_set(LED_2, 0.0, 0);
    led_set(LED_3, 0.0, 0);
    led_set(LED_4, 0.0, 0);
}
```

closed 2019-06-16 by J2Kbr

AJSinc, 2019-06-18:
Your code causes the LED to blink too, which I believe is not intended.