J2Kbr/TitanTwo #209: Issues with persistently pairing joycons to titantwo

esotericist, 2019-05-31:
I'm using a titantwo on a mac running mojave (10.14), and I'm having a great deal of difficulty getting joycons to pair with the titantwo reliably.

I can use the "pair devices" button in the "device configuration" panel of the gtunerIV app, but this doesn't seem to save any kind of pairing information so it doesn't help me when I have the titantwo connected via the "output" port instead of the "prog" port.

If I hold "down" on the titantwo to initiate pairing, I can reliably get *one* of the two joycons to pair, but getting the second joycon to pair requires a great deal of attempts before I eventually get them both. And then, for a span of time, it works.

But as soon as the joycons are turned off, it seems like the pairing is forgotten, and I have to go through the same procedure all over again, typically taking between ten and thirty minutes of fussing over holding buttons (both on the titantwo and on the joycons) before I get them both paired again.

I am being careful to turn on the joycons using the "home" and "capture" buttons as the documentation I could find advises, but this seems to make no difference on whether or not the titantwo recognizes that the joycon has previously paired with it.

esotericist, 2019-06-04:
So after working with someone in the consoletuner discord, the problem has been clarified as not "joycons are no longer paired with the titantwo", but instead "joycons frequently take far too long to sync, giving the impression of no longer being paired with the titantwo".

Using the trick of canceling a failed sync via turning the joycons off with tapping the sync button so you can attempt a sync again helps, but it can still take me as many as a couple dozen attempts with each joycon of a set before I'm able to use them together, which is definitely not ideal.

J2Kbr, 2019-06-16:
Firmware 1.03 reverts the page scan parameters change made in v1.01 and sets page scan mode to interlaced. In my tests these changes improved connection speed of the Joycons.

esotericist, 2019-06-16:
Is this firmware available yet? I can only see 1.02 in the downloads page for Titan Two Firmware.

J2Kbr, 2019-06-16:
FW 1.03 will be released early this week. thanks.

esotericist, 2020-01-19:
@J2Kbr 
Just for the record, I have continued to have this problem, even up to the latest firmware update (updated just now, so presumably it's the one published in discord as 2020-01-18).

if i can manage to get both joycons connected to the titan two at once, they'll work fine for weeks but it takes countless connect attempts to get them both simultaneously connected. this remains very vexing.

J2Kbr, 2020-03-03:
Investigating other Bluetooth related issue it was verified, under certain loads, the communication with the Bluetooth module could result in corrupted data. The issue was fixed (FW 1.07, to be released this week) and has potential to solve the dual Joycon pairing/connection problem.

In case the problem persists, with this update Bluetooth errors are now reported on Gtuner's output panel. Please copy and paste any printed error(s) while trying connect the Joycons to the Titan Two, I will then reopen this issue and we move from there.

Thank you.

esotericist, 2020-03-09:
I just installed the Gtuner IV 1.7 RC, and enabled the "beta tester" flag in the config in order to install the new firmware (1.7.2.0 according to the gtuner IV interface).

I am still experiencing essentially the same problems as before: it is difficult to get both joycons connected to the titantwo at the same time. once i do, they tend to stay that way as long as they don't get turned off, but it's still an onerous process.

The information from the log:
```
Device Selector: added 1 device(s), total of 1 device(s) detected
Connect: TITAN TWO [E936FFFFEBBB23543953343638362020FF0D0BF5] connected (Power-on)
Micro-SD Card: Inserted (481.00MB)
Bluetooth Module: Device disconnection [ evt 5, err 0, info 010016 ]
Bluetooth Module: Connection error [ evt 15, err 2, info 010604 ]
```

event sequencing for the above log:
1. plug in titantwo on the 'prog' port (so that gtuner can give diagnostic information)
2. press the 'home' button on the right joycon, let it connect. This is when the "Device disconnection" line happens, at the moment of successful connection.
3. press the 'capture' button on the left joycon. it never connects.

i get essentially the same results if I swap the sequencing of which joycon I activate first, just the first one to connect is the one that connects. And the log is slightly different:
```
Bluetooth Module: Device disconnection [ evt 5, err 0, info 010013 ]
```
I don't consistently get the "connection error" message, but I do consistently get the "device disconnection" message when a joycon successfully connects.

i will say it is slightly more willing to let me successfully force-pair the joycons than it used to be when not connected on the 'prog' port; if I let one joycon connect, put the titantwo into pairing mode without turning off the first joycon, it is much more likely to accept the pairing from the second joycon now. (not guaranteed, but likely).  

The problem here is that i am physically impaired (i use joycons for gaming as an accessibility aide), and holding down the button on the titantwo in order to put it into pairing mode when it's not connected to the 'prog' port is genuinely painful for me. (and trying to press the pair button on the joycons isn't easy either)

also vexingly, it is *more* stubborn when connected on the 'prog' port than it used to be. it is in fact easier for me to get it working now when on the 'output' port.

Example log of trying to force pair both joycons using the gtuner interface:
```
Device Selector: added 1 device(s), total of 1 device(s) detected
Connect: TITAN TWO [E936FFFFEBBB23543953343638362020FF0D0BF5] connected (Power-on)
Micro-SD Card: Inserted (481.00MB)
Bluetooth Module: Wireless Bluetooth Pairing: put the controller or console in pairing mode within 20 seconds
Bluetooth Module: Wireless Bluetooth Pairing: put the controller or console in pairing mode within 20 seconds
Bluetooth Module: Device disconnection [ evt 5, err 0, info 020008 ]
Bluetooth Module: Device disconnection [ evt 5, err 0, info 010008 ]
Bluetooth Module: Connection error [ evt 15, err 2, info 010604 ]
Disconnect: Titan Two device disconnected (USB Disconnection)
```
notably, there was a minute delay between the attempt at pairing the second joycon and the disconnection messages.
[image]

i hope any of this information is useful.

J2Kbr, 2020-03-09:
Thank you, the provided information will be very helpful to debug this issue.

esotericist, 2020-03-09:
Let me know if there's any more targeted tests that might give any additional information needed. I'm willing to do my part to help. :)

J2Kbr, 2020-05-08:
During the past days I've been investigating this issue and verified the Joycon invalidates the paired pair if, for any reason, the attempt of connection fails more than 2 or 3 times. I also noticed, if a Joycon is not used for a certain period of time, the pairing no longer works.

Despite efforts of sniff the Joycon Bluetooth communication, nothing relevant to this issue could be found. The logs only shows regular Bluetooth data.

However, I believe to have found a reliable way to pair two Joycons with the Titan Two.

Start with the left Joycon. Put the Titan Two in Bluetooth pairing mode and press the pairing button on the Joycon. Wait for the pairing/connection succeed and, after that, turn off the Joycon. Press the capture button on the left Joycon for it connect to the Titan Two, wait until the connection is established. Keep the left Joycon connected.

Now the right Joycon. Put the Titan Two in Bluetooth pairing mode and press the pairing button on the Joycon. Both Joycons should now be paired and subsequent connections should work just fine, provided the Joycos are used with the titan Two on a regular basis. Otherwise this pairing procedure must be accomplished again.

Hope this helps.

closed 2020-05-08 by J2Kbr