J2Kbr/TitanTwo #210: Astro C40 Stopped working after firmware 1.3.3.0

psawjack, 2019-06-26:
The previous firmware version was working with the modification to the sdconfig dir but  after update no longer works. Attempt was made to reconfigure the SD card with extra PS4 file like before but it didn't fix issue.

J2Kbr, 2019-08-20:
I am currently working on this issue, could you please capture the USB data traffic of Astro C40 with the PS4 console?

1) Unplug all controllers from the Titan Two;
2) Connect the output to the PS4 console;
3) Connect the Prog port to your computer;
4) Open Gtuner IV and go to top menu Tools -> "Fetch Descriptors ...";
5) Plug the Astro C40 to any port at the front of the Titan Two;
6) Wait couple seconds for this window be update with the captured data;
7) Click on the button "Copy Descriptors" and send me.

psawjack, 2019-08-20:
I cannot help tight now am currently on holiday for another week. I will attempt to follow up after I get back.

psawjack, 2019-08-30:
[USB DESCRIPTOR]
12 01 00 02 00 00 00 40 86 98 25 00 71 00 01 02 03 01 09 02 80 01 05 01 00 80 FA 09 04 00 00 00 01 01 00 00 0A 24 01 00 01 34 00 02 01 02 0C 24 02 01 02 04 00 01 00 00 00 04 09 24 03 04 01 01 00 01 04 0C 24 02 05 01 01 00 02 03 00 00 05 09 24 03 09 02 04 00 05 05 09 04 01 00 00 01 02 00 00 09 04 01 01 01 01 02 00 00 07 24 01 04 01 01 00 1A 24 02 01 01 02 10 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 83 0D 62 00 01 00 00 07 25 01 01 01 01 00 09 04 01 02 01 01 02 00 00 07 24 01 04 01 01 00 1A 24 02 01 01 03 18 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 83 0D 93 00 01 00 00 07 25 01 01 01 01 00 09 04 02 00 00 01 02 00 00 09 04 02 01 01 01 02 00 00 07 24 01 05 01 01 00 1A 24 02 01 02 02 10 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 03 0D C4 00 01 00 00 07 25 01 01 01 01 00 09 04 02 02 01 01 02 00 00 07 24 01 05 01 01 00 1A 24 02 01 02 03 18 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 03 0D 26 01 01 00 00 07 25 01 01 01 01 00 09 04 03 00 02 03 00 00 00 09 21 11 01 00 01 22 4C 00 07 05 81 03 40 00 01 07 05 01 03 40 00 01 09 04 04 00 02 03 00 00 00 09 21 11 01 00 01 22 AB 00 07 05 82 03 40 00 05 07 05 02 03 40 00 05

[STRING 1 DESCRIPTOR]
00 00 00 03 00 00 02 01 80 06 03 01 04 09 00 1A
1A 03 41 00 53 00 54 00 52 00 4F 00 20 00 47 00 61 00 6D 00 69 00 6E 00 67 00

[STRING 2 DESCRIPTOR]
00 00 00 03 00 00 02 01 80 06 03 02 04 09 00 14
14 03 41 00 53 00 54 00 52 00 4F 00 20 00 43 00 34 00 30 00

[STRING 3 DESCRIPTOR]
00 00 00 03 00 00 02 01 80 06 03 03 04 09 00 16
16 03 66 00 66 00 30 00 30 00 30 00 31 00 30 00 65 00 36 00 39 00

[INTERFACE 3 HID DESCRIPTOR]
00 00 00 03 00 00 02 01 81 06 22 00 00 03 00 4C
06 32 FF 09 74 A1 01 85 41 09 75 15 00 26 FF 00 95 3F 75 08 81 02 85 41 09 76 15 00 26 FF 00 95 3F 75 08 91 02 C0 06 58 FF 09 01 A1 01 85 58 09 01 15 00 26 FF 00 95 3F 75 08 81 02 85 58 09 01 15 00 26 FF 00 95 3F 75 08 91 02 C0

[INTERFACE 4 HID DESCRIPTOR]
00 00 00 03 00 00 02 01 81 06 22 00 00 04 00 AB
05 01 09 05 A1 01 85 01 09 30 09 31 09 32 09 35 15 00 26 FF 00 75 08 95 04 81 02 09 39 15 00 25 07 35 00 46 3B 01 65 14 75 04 95 01 81 42 65 00 05 09 19 01 29 0E 15 00 25 01 75 01 95 0E 81 02 06 00 FF 09 20 75 06 95 01 81 02 05 01 09 33 09 34 15 00 26 FF 00 75 08 95 02 81 02 06 00 FF 09 21 95 36 81 02 85 05 09 22 95 1F 91 02 85 03 0A 21 27 95 2F B1 02 06 80 FF 85 E0 09 57 95 02 B1 02 C0 06 F0 FF 09 40 A1 01 85 F0 09 47 95 3F B1 02 85 F1 09 48 95 3F B1 02 85 F2 09 49 95 0F B1 02 85 F3 0A 01 47 95 07 B1 02 C0

[USB SNIFF]
00 00 00 01 00 00 01 00 00 05 00 11 00 00 00 00
00 00 00 03 00 00 02 01 00 05 00 04 00 00 00 00
11
00 00 00 01 00 00 01 00 80 06 01 00 00 00 00 08
00 00 00 03 00 00 02 01 80 06 01 00 00 00 00 08
12 01 00 02 00 00 00 40
00 00 00 01 00 00 01 00 80 06 01 00 00 00 00 12
00 00 00 03 00 00 02 01 80 06 01 00 00 00 00 12
12 01 00 02 00 00 00 40 86 98 25 00 71 00 01 02 03 01
00 00 00 01 00 00 01 00 80 06 03 00 00 00 00 02
00 00 00 03 00 00 02 01 80 06 03 00 00 00 00 02
04 03
00 00 00 01 00 00 01 00 80 06 03 00 00 00 00 04
00 00 00 03 00 00 02 01 80 06 03 00 00 00 00 04
04 03 09 04
00 00 00 01 00 00 01 00 80 06 03 03 04 09 00 02
00 00 00 03 00 00 02 01 80 06 03 03 04 09 00 02
16 03
00 00 00 01 00 00 01 00 80 06 03 03 04 09 00 16
00 00 00 03 00 00 02 01 80 06 03 03 04 09 00 16
16 03 66 00 66 00 30 00 30 00 30 00 31 00 30 00 65 00 36 00 39 00
00 00 00 01 00 00 01 00 80 06 03 01 04 09 00 02
00 00 00 03 00 00 02 01 80 06 03 01 04 09 00 02
1A 03
00 00 00 01 00 00 01 00 80 06 03 01 04 09 00 1A
00 00 00 03 00 00 02 01 80 06 03 01 04 09 00 1A
1A 03 41 00 53 00 54 00 52 00 4F 00 20 00 47 00 61 00 6D 00 69 00 6E 00 67 00
00 00 00 01 00 00 01 00 80 06 03 02 04 09 00 02
00 00 00 03 00 00 02 01 80 06 03 02 04 09 00 02
14 03
00 00 00 01 00 00 01 00 80 06 03 02 04 09 00 14
00 00 00 03 00 00 02 01 80 06 03 02 04 09 00 14
14 03 41 00 53 00 54 00 52 00 4F 00 20 00 43 00 34 00 30 00
00 00 00 01 00 00 01 00 80 06 02 00 00 00 00 09
00 00 00 03 00 00 02 01 80 06 02 00 00 00 00 09
09 02 80 01 05 01 00 80 FA
00 00 00 01 00 00 01 00 80 06 02 00 00 00 01 80
00 00 00 03 00 00 02 01 80 06 02 00 00 00 01 80
09 02 80 01 05 01 00 80 FA 09 04 00 00 00 01 01 00 00 0A 24 01 00 01 34 00 02 01 02 0C 24 02 01 02 04 00 01 00 00 00 04 09 24 03 04 01 01 00 01 04 0C 24 02 05 01 01 00 02 03 00 00 05 09 24 03 09 02 04 00 05 05 09 04 01 00 00 01 02 00 00 09 04 01 01 01 01 02 00 00 07 24 01 04 01 01 00 1A 24 02 01 01 02 10 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 83 0D 62 00 01 00 00 07 25 01 01 01 01 00 09 04 01 02 01 01 02 00 00 07 24 01 04 01 01 00 1A 24 02 01 01 03 18 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 83 0D 93 00 01 00 00 07 25 01 01 01 01 00 09 04 02 00 00 01 02 00 00 09 04 02 01 01 01 02 00 00 07 24 01 05 01 01 00 1A 24 02 01 02 02 10 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 03 0D C4 00 01 00 00 07 25 01 01 01 01 00 09 04 02 02 01 01 02 00 00 07 24 01 05 01 01 00 1A 24 02 01 02 03 18 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 03 0D 26 01 01 00 00 07 25 01 01 01 01 00 09 04 03 00 02 03 00 00 00 09 21 11 01 00 01 22 4C 00 07 05 81 03 40 00 01 07 05 01 03 40 00 01 09 04 04 00 02 03 00 00 00 09 21 11 01 00 01 22 AB 00 07 05 82 03 40 00 05 07 05 02 03 40 00 05
00 00 00 01 00 00 01 00 00 09 00 01 00 00 00 00
00 00 00 03 00 00 02 01 00 09 00 01 00 00 00 00
00 00 00 01 00 00 01 00 01 0B 00 02 00 02 00 00
00 00 00 03 00 00 02 01 01 0B 00 02 00 02 00 00
[... cut ...]
01 73 83 9A 90 08 00 01 00 00 AD 77 00 14 00 0D 00 17 00 30 04 E1 1F 2D 05 00 00 00 00 00 00 00 00 00 00 80 FF FF FF 80 FF FF FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 03 00 02
01 73 83 9A 90 08 00 01 00 00 FB 77 00 13 00 0D 00 1E 00 F5 03 41 20 DE 04 00 00 00 00 00 00 00 00 00 00 80 FF FF FF 80 FF FF FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 01 00 02
05 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

psawjack, 2019-08-30:
Here is the info. Thanks.

J2Kbr, 2019-09-03:
Thank you for sending the complete descriptors. The C40 should fully work with the Titan Two firmware 1.05 (to be released very soon).

adool711, 2019-09-04:
I still can't run the Astro C40 controller on Titan Two

[USB DESCRIPTOR]
12 01 00 02 00 00 00 40 86 98 25 00 71 00 01 02 03 01 09 02 80 01 05 01 00 80 FA 09 04 00 00 00 01 01 00 00 0A 24 01 00 01 34 00 02 01 02 0C 24 02 01 02 04 00 01 00 00 00 04 09 24 03 04 01 01 00 01 04 0C 24 02 05 01 01 00 02 03 00 00 05 09 24 03 09 02 04 00 05 05 09 04 01 00 00 01 02 00 00 09 04 01 01 01 01 02 00 00 07 24 01 04 01 01 00 1A 24 02 01 01 02 10 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 83 0D 62 00 01 00 00 07 25 01 01 01 01 00 09 04 01 02 01 01 02 00 00 07 24 01 04 01 01 00 1A 24 02 01 01 03 18 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 83 0D 93 00 01 00 00 07 25 01 01 01 01 00 09 04 02 00 00 01 02 00 00 09 04 02 01 01 01 02 00 00 07 24 01 05 01 01 00 1A 24 02 01 02 02 10 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 03 0D C4 00 01 00 00 07 25 01 01 01 01 00 09 04 02 02 01 01 02 00 00 07 24 01 05 01 01 00 1A 24 02 01 02 03 18 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 03 0D 26 01 01 00 00 07 25 01 01 01 01 00 09 04 03 00 02 03 00 00 00 09 21 11 01 00 01 22 4C 00 07 05 81 03 40 00 01 07 05 01 03 40 00 01 09 04 04 00 02 03 00 00 00 09 21 11 01 00 01 22 AB 00 07 05 82 03 40 00 05 07 05 02 03 40 00 05 

[STRING 1 DESCRIPTOR]
00 00 00 03 00 00 02 01 80 06 03 01 04 09 00 1A 
1A 03 41 00 53 00 54 00 52 00 4F 00 20 00 47 00 61 00 6D 00 69 00 6E 00 67 00 

[STRING 2 DESCRIPTOR]
00 00 00 03 00 00 02 01 80 06 03 02 04 09 00 14 
14 03 41 00 53 00 54 00 52 00 4F 00 20 00 43 00 34 00 30 00 

[STRING 3 DESCRIPTOR]
00 00 00 03 00 00 02 01 80 06 03 03 04 09 00 16 
16 03 66 00 66 00 30 00 30 00 30 00 31 00 63 00 62 00 33 00 37 00 

[INTERFACE 3 HID DESCRIPTOR]
00 00 00 03 00 00 02 01 81 06 22 00 00 03 00 4C 
06 32 FF 09 74 A1 01 85 41 09 75 15 00 26 FF 00 95 3F 75 08 81 02 85 41 09 76 15 00 26 FF 00 95 3F 75 08 91 02 C0 06 58 FF 09 01 A1 01 85 58 09 01 15 00 26 FF 00 95 3F 75 08 81 02 85 58 09 01 15 00 26 FF 00 95 3F 75 08 91 02 C0 

[INTERFACE 4 HID DESCRIPTOR]
00 00 00 03 00 00 02 01 81 06 22 00 00 04 00 AB 
05 01 09 05 A1 01 85 01 09 30 09 31 09 32 09 35 15 00 26 FF 00 75 08 95 04 81 02 09 39 15 00 25 07 35 00 46 3B 01 65 14 75 04 95 01 81 42 65 00 05 09 19 01 29 0E 15 00 25 01 75 01 95 0E 81 02 06 00 FF 09 20 75 06 95 01 81 02 05 01 09 33 09 34 15 00 26 FF 00 75 08 95 02 81 02 06 00 FF 09 21 95 36 81 02 85 05 09 22 95 1F 91 02 85 03 0A 21 27 95 2F B1 02 06 80 FF 85 E0 09 57 95 02 B1 02 C0 06 F0 FF 09 40 A1 01 85 F0 09 47 95 3F B1 02 85 F1 09 48 95 3F B1 02 85 F2 09 49 95 0F B1 02 85 F3 0A 01 47 95 07 B1 02 C0 

[USB SNIFF]
00 00 00 01 00 00 01 00 00 05 00 18 00 00 00 00 
00 00 00 03 00 00 02 01 00 05 00 04 00 00 00 00 
18 
00 00 00 01 00 00 01 00 80 06 01 00 00 00 00 08 
00 00 00 03 00 00 02 01 80 06 01 00 00 00 00 08 
12 01 00 02 00 00 00 40 
00 00 00 01 00 00 01 00 80 06 01 00 00 00 00 12 
00 00 00 03 00 00 02 01 80 06 01 00 00 00 00 12 
12 01 00 02 00 00 00 40 86 98 25 00 71 00 01 02 03 01 
00 00 00 01 00 00 01 00 80 06 03 00 00 00 00 02 
00 00 00 03 00 00 02 01 80 06 03 00 00 00 00 02 
04 03 
00 00 00 01 00 00 01 00 80 06 03 00 00 00 00 04 
00 00 00 03 00 00 02 01 80 06 03 00 00 00 00 04 
04 03 09 04 
00 00 00 01 00 00 01 00 80 06 03 03 04 09 00 02 
00 00 00 03 00 00 02 01 80 06 03 03 04 09 00 02 
16 03 
00 00 00 01 00 00 01 00 80 06 03 03 04 09 00 16 
00 00 00 03 00 00 02 01 80 06 03 03 04 09 00 16 
16 03 66 00 66 00 30 00 30 00 30 00 31 00 63 00 62 00 33 00 37 00 
00 00 00 01 00 00 01 00 80 06 03 01 04 09 00 02 
00 00 00 03 00 00 02 01 80 06 03 01 04 09 00 02 
1A 03 
00 00 00 01 00 00 01 00 80 06 03 01 04 09 00 1A 
00 00 00 03 00 00 02 01 80 06 03 01 04 09 00 1A 
1A 03 41 00 53 00 54 00 52 00 4F 00 20 00 47 00 61 00 6D 00 69 00 6E 00 67 00 
00 00 00 01 00 00 01 00 80 06 03 02 04 09 00 02 
00 00 00 03 00 00 02 01 80 06 03 02 04 09 00 02 
14 03 
00 00 00 01 00 00 01 00 80 06 03 02 04 09 00 14 
00 00 00 03 00 00 02 01 80 06 03 02 04 09 00 14 
14 03 41 00 53 00 54 00 52 00 4F 00 20 00 43 00 34 00 30 00 
00 00 00 01 00 00 01 00 80 06 02 00 00 00 00 09 
00 00 00 03 00 00 02 01 80 06 02 00 00 00 00 09 
09 02 80 01 05 01 00 80 FA 
00 00 00 01 00 00 01 00 80 06 02 00 00 00 01 80 
00 00 00 03 00 00 02 01 80 06 02 00 00 00 01 80 
09 02 80 01 05 01 00 80 FA 09 04 00 00 00 01 01 00 00 0A 24 01 00 01 34 00 02 01 02 0C 24 02 01 02 04 00 01 00 00 00 04 09 24 03 04 01 01 00 01 04 0C 24 02 05 01 01 00 02 03 00 00 05 09 24 03 09 02 04 00 05 05 09 04 01 00 00 01 02 00 00 09 04 01 01 01 01 02 00 00 07 24 01 04 01 01 00 1A 24 02 01 01 02 10 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 83 0D 62 00 01 00 00 07 25 01 01 01 01 00 09 04 01 02 01 01 02 00 00 07 24 01 04 01 01 00 1A 24 02 01 01 03 18 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 83 0D 93 00 01 00 00 07 25 01 01 01 01 00 09 04 02 00 00 01 02 00 00 09 04 02 01 01 01 02 00 00 07 24 01 05 01 01 00 1A 24 02 01 02 02 10 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 03 0D C4 00 01 00 00 07 25 01 01 01 01 00 09 04 02 02 01 01 02 00 00 07 24 01 05 01 01 00 1A 24 02 01 02 03 18 06 80 3E 00 22 56 00 C0 5D 00 00 7D 00 44 AC 00 80 BB 00 09 05 03 0D 26 01 01 00 00 07 25 01 01 01 01 00 09 04 03 00 02 03 00 00 00 09 21 11 01 00 01 22 4C 00 07 05 81 03 40 00 01 07 05 01 03 40 00 01 09 04 04 00 02 03 00 00 00 09 21 11 01 00 01 22 AB 00 07 05 82 03 40 00 05 07 05 02 03 40 00 05 
00 00 00 01 00 00 01 00 00 09 00 01 00 00 00 00 
[... cut ...]
33 FC FF FE 00 00 AC 04 00 49 02 00 E8 02 00 40 04 00 3B FE FF 55 FE FF 80 00 00 68 00 00 B2 00 00 A9 FC FF 08 01 00 C3 02 00 AA 03 00 97 00 00 3D FF FF D8 02 00 4D 01 00 07 04 00 39 FF FF DF 00 00 C0 01 00 87 FF FF 76 01 00 FC FD FF 32 01 00 A4 00 00 BA FE FF 80 FF FF 60 FD FF 9A FF FF BB FF FF CE FD FF F8 FC FF 91 FB FF 7D FE FF E4 FD FF C1 FE FF FA FF FF 5D FB FF 2A FD FF F7 FE FF 
00 00 00 01 00 02 
05 81 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 4D 00 00 00 00 00 00 00 00 00

J2Kbr, 2019-09-04:
Please confirm to me if you tried with the firmware 1.05-beta, as explained here:

https://www.consoletuner.com/forum/viewtopic.php?f=2&t=13510

Thank you.

adool711, 2019-09-04:
@J2Kbr Yes, I am talking after I try beta 1.05, but I still suffer that the PlayStation 4 does not respond to the commands when connecting the Astro C40 controller to two Titan

J2Kbr, 2019-09-04:
thank you, reopening this issue.

J2Kbr, 2019-09-05:
https://www.consoletuner.com/forum/viewtopic.php?f=23&t=13510

> 1) Does the status LED of Input-A is green? as well as for the output status LED?
> Answer is yes...
> 2) On Device Monitor is possible see the controller inputs being activated as normal?
> Answer is no while output to ps4 is on but yes when output is off.
> But wired mode presses the ps button at full press and holds there true and only that button by its self.
> Wireless mode does nothing when output is on. When output is off works as it should.
> 
> Ok I checked it out and played around with it all.
> Yes both colors are green on A slot and output but when you get to this point the device monitor
> Show the ps button being pressed at 100 percent and freezes there.
> But if you don't hook up to ps4 with output the device monitor works as it should all buttons work correctly.
> I don't know if this is a issue inreguards to the fact you can not turn your ps4 on directly with c40 ps button. But seems weird that when connecting properly device monitor shows ps button being stuck at 100 percent press.
> 
> Also the c40 works with otg in program slot but seeks for authorization and times out at 10 minutes with touchpad and share button being swapped..
> 
> WIRELESS MODE WITH C40 BLUETOOTH DONGLE BELOW.
> Similar results both colors are green but device monitor does nothing it does not show ps button being pressed like
> Wired mode does.

J2Kbr, 2019-09-10:
Soon (today/tomorrow) I am releasing a new 1.05B revision with a change in the PS4 emulator, not sure if it will fix the issue with C40, but worth the try. Please keep me updated on the results after update your Titan Two. thanks.

closed 2019-09-10 by J2Kbr

J2Kbr, 2019-09-11:
```
[USB DESCRIPTOR]
12           bLength: 18
01           bDescriptorType: 1 - DEVICE
00 02        bcdUSB: 2.00
00           bDeviceClass: 0 (00)
00           bDeviceSubClass: 0 (00)
00           bDeviceProtocol: 0 (00)
40           bMaxPacketSize0: 64
86 98        idVendor: 9886
25 00        idProduct: 0025
71 00        bcdDevice: 0.71
01           iManufacturer: 1
02           iProduct: 2
03           iSerialNumber: 3
01           bNumConfigurations: 1

09           bLength: 9
02           bDescriptorType: 2 - CONFIGURATION
80 01        wTotalLength: 384
05           bNumInterfaces: 5
01           bConfigurationValue: 1
00           iConfiguration: 0
80           bmAttributes: 128 (Bus Powered)
FA           MaxPower: 500mA

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
00             bInterfaceNumber: 0
00             bAlternateSetting: 0
00             bNumEndpoints: 0
01             bInterfaceClass: 1 (01)
01             bInterfaceSubClass: 1 (01)
00             bInterfaceProtocol: 0 (00)
00             iInterface: 0

0A               bLength: 10
24               bDescriptorType: 36 - UNKNOW (0A 24 01 00 01 34 00 02 01 02)

0C               bLength: 12
24               bDescriptorType: 36 - UNKNOW (0C 24 02 01 02 04 00 01 00 00 00 04)

09               bLength: 9
24               bDescriptorType: 36 - UNKNOW (09 24 03 04 01 01 00 01 04)

0C               bLength: 12
24               bDescriptorType: 36 - UNKNOW (0C 24 02 05 01 01 00 02 03 00 00 05)

09               bLength: 9
24               bDescriptorType: 36 - UNKNOW (09 24 03 09 02 04 00 05 05)

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
01             bInterfaceNumber: 1
00             bAlternateSetting: 0
00             bNumEndpoints: 0
01             bInterfaceClass: 1 (01)
02             bInterfaceSubClass: 2 (02)
00             bInterfaceProtocol: 0 (00)
00             iInterface: 0

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
01             bInterfaceNumber: 1
01             bAlternateSetting: 1
01             bNumEndpoints: 1
01             bInterfaceClass: 1 (01)
02             bInterfaceSubClass: 2 (02)
00             bInterfaceProtocol: 0 (00)
00             iInterface: 0
[... 278 lines cut ...]
B1 02          (MAIN)   FEATURE            0x00000002 (2 fields x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
C0           (MAIN)   END_COLLECTION     Application

06 F0FF      (GLOBAL) USAGE_PAGE         0xFFF0 Vendor-defined 
09 40        (LOCAL)  USAGE              0xFFF00040  
A1 01        (MAIN)   COLLECTION         0x01 Application (Usage=0xFFF00040: Page=Vendor-defined, Usage=, Type=)
85 F0          (GLOBAL) REPORT_ID          0xF0 (240) 
09 47          (LOCAL)  USAGE              0xFFF00047  
95 3F          (GLOBAL) REPORT_COUNT       0x3F (63) Number of fields 
B1 02          (MAIN)   FEATURE            0x00000002 (63 fields x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
85 F1          (GLOBAL) REPORT_ID          0xF1 (241) 
09 48          (LOCAL)  USAGE              0xFFF00048  
95 3F          (GLOBAL) REPORT_COUNT       0x3F (63) Number of fields
B1 02          (MAIN)   FEATURE            0x00000002 (63 fields x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
85 F2          (GLOBAL) REPORT_ID          0xF2 (242) 
09 49          (LOCAL)  USAGE              0xFFF00049  
95 0F          (GLOBAL) REPORT_COUNT       0x0F (15) Number of fields 
B1 02          (MAIN)   FEATURE            0x00000002 (15 fields x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
85 F3          (GLOBAL) REPORT_ID          0xF3 (243) 
0A 0147        (LOCAL)  USAGE              0xFFF04701  
95 07          (GLOBAL) REPORT_COUNT       0x07 (7) Number of fields 
B1 02          (MAIN)   FEATURE            0x00000002 (7 fields x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
C0           (MAIN)   END_COLLECTION     Application

```

adool711, 2019-09-16:
Hi Is there anyone who uses Astro C40 controller and has downloaded Beta 1.5.3.0 and starts noticing a slow response because I downloaded the update as soon as it came down but I didn't notice any improvement or misuse of my use and this day I started to feel slow in responding to the orders which makes me feel at playing Fort Knight Heavy weight during gameplay and slow response especially by jogging is evident to me

J2Kbr, 2019-09-16:
Hi adool, in the Device Configurations, try set the Input Polling Rate to 1000Hz. Thanks.