J2Kbr/TitanTwo #212: Feature Request T2: Controller Headset Support Over Bluetooth For Consoles

jellybean44, 2019-06-29:
**Feature Request T2: Controller Headset Support Over Bluetooth For Consoles**

Titan Two's users who purchased the Titan Expansion Kit are not currently able to fully leverage the capabilities of the Bluetooth Module. End-users who use his/her headsets connected to a controller on PS4 or XBOX are unable to do so while having the Titan Two paired to the controller wirelessly. At the moment users need to connected the devices in full-passthru mode. 

The request is to implement controller headset support over bluetooth for consoles so that users who connect his/her headsets directly to the controller can also pair the controller to the Titan Two device wirelessly ; therefore, not having to connect the devices in full-passthru mode.

AJSinc, 2019-06-30:
The Xbox doesn't support headsets when connected with BT to any device.

jellybean44, 2019-06-30:
Thank you, AJSinc. I wasn't aware of that. I will edit my post so that only PS4 is in scope.

J2Kbr, 2020-09-23:
Firmware 1.08 (currently in beta testing) adds support for controller headsets over wireless connection for Xbox One controller via Xbox Wireless Receiver when using on Xbox One console.

After investigation and researching, it was verified the audio data over Bluetooth connection of PS4 controllers is not PCM. The audio data is compressed with an unknown format. Unfortunately this prevents add support of headsets for PS4 controllers when connected over Bluetooth (via USB is already supported). Even if the compression format is know, the Titan Two does not have the necessarily hardware to handle the audio decompression and conversion to PCM.

closed 2020-09-23 by J2Kbr