J2Kbr/TitanTwo #216: Add support for pokken controller for Wii U/Switch

J2Kbr, 2019-07-24:
> I just got a pokken controller for Wii U/Switch. It is having trouble connecting to the Titan Two (red light by the input). However, if I plug in two at the same time, the second one connects. Here are the descriptors if this helps.

```
[USB DESCRIPTOR]
12 01 00 02 00 00 00 40 0D 0F 92 00 00 01 01 02 00 01 09 02 29 00 01 01 00 80 FA 09 04 00 00 02 03 00 00 00 09 21 11 01 00 01 22 5A 00 07 05 02 03 40 00 05 07 05 81 03 40 00 05 

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 1C 
1C 03 48 00 4F 00 52 00 49 00 20 00 43 00 4F 00 2E 00 2C 00 4C 00 54 00 44 00 2E 00 

[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 24 
24 03 50 00 4F 00 4B 00 4B 00 45 00 4E 00 20 00 43 00 4F 00 4E 00 54 00 52 00 4F 00 4C 00 4C 00 45 00 52 00 

[INTERFACE 0 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 00 00 5A 
05 01 09 05 A1 01 15 00 25 01 35 00 45 01 75 01 95 0D 05 09 19 01 29 0D 81 02 95 03 81 01 05 01 25 07 46 3B 01 75 04 95 01 65 14 09 39 81 42 65 00 95 01 81 01 26 FF 00 46 FF 00 09 30 09 31 09 32 09 35 75 08 95 04 81 02 06 00 FF 09 20 95 01 81 02 0A 21 26 95 08 91 02 C0

```

J2Kbr, 2019-08-20:
After checking the descriptors it was verified this controller should be fully supported by the Titan Two. The described issue seems to be related with electrical connection.

closed 2019-08-20 by J2Kbr

thomasmasonnelson, 2019-09-06:
The input lights are still red at first, shown in this video: [https://photos.app.goo.gl/U8bFA4owfs3JYrK67](url)