J2Kbr/TitanTwo #217: Add support for mayflash gamecube controller converter for Wii U/Switch/PC

J2Kbr, 2019-07-24:
> I also got a mayflash gamecube controller converter for Wii U/Switch/PC. The input light is green but no inputs are read in the device monitor. Here are the descriptors if this helps.

```
[USB DESCRIPTOR]
12 01 00 02 00 00 00 40 7E 05 37 03 00 01 01 02 03 01 09 02 29 00 01 01 00 E0 FA 09 04 00 00 02 03 00 00 00 09 21 10 01 00 01 22 D6 00 07 05 81 03 25 00 08 07 05 02 03 05 00 08 

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 12 
12 03 4E 00 69 00 6E 00 74 00 65 00 6E 00 64 00 6F 00 

[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 10 
10 03 57 00 55 00 50 00 2D 00 30 00 32 00 38 00 

[STRING 3 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 03 04 09 00 16 
16 03 31 00 35 00 2F 00 30 00 37 00 2F 00 32 00 30 00 31 00 34 00 

[INTERFACE 0 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 00 00 D6 
05 05 09 00 A1 01 85 11 19 00 2A FF 00 15 00 26 FF 00 75 08 95 05 91 00 C0 A1 01 85 21 19 00 2A FF 00 15 00 26 FF 00 75 08 95 25 81 00 C0 A1 01 85 12 19 00 2A FF 00 15 00 26 FF 00 75 08 95 01 91 00 C0 A1 01 85 22 19 00 2A FF 00 15 00 26 FF 00 75 08 95 19 81 00 C0 A1 01 85 13 19 00 2A FF 00 15 00 26 FF 00 75 08 95 01 91 00 C0 A1 01 85 23 19 00 2A FF 00 15 00 26 FF 00 75 08 95 02 81 00 C0 A1 01 85 14 19 00 2A FF 00 15 00 26 FF 00 75 08 95 01 91 00 C0 A1 01 85 24 19 00 2A FF 00 15 00 26 FF 00 75 08 95 02 81 00 C0 A1 01 85 15 19 00 2A FF 00 15 00 26 FF 00 75 08 95 01 91 00 C0 A1 01 85 25 19 00 2A FF 00 15 00 26 FF 00 75 08 95 02 81 00 C0

```

J2Kbr, 2019-08-20:
Does this descriptors were captured in `PC Mode` or `WiiU/NS` mode?

thomasmasonnelson, 2019-09-06:
PC Mode:
`[USB DESCRIPTOR]
12 01 00 02 00 00 00 40 79 00 47 18 00 01 01 02 00 01 09 02 3B 00 02 01 00 C0 FA 09 04 00 00 01 03 00 00 00 09 21 10 01 00 01 22 61 00 07 05 81 03 40 00 04 09 04 01 00 01 03 00 00 00 09 21 10 01 00 01 22 61 00 07 05 82 03 40 00 04 

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 22 
22 03 6D 00 61 00 79 00 66 00 6C 00 61 00 73 00 68 00 20 00 6C 00 69 00 6D 00 69 00 74 00 65 00 64 00 

[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 4A 
4A 03 4D 00 41 00 59 00 46 00 4C 00 41 00 53 00 48 00 20 00 47 00 61 00 6D 00 65 00 43 00 75 00 62 00 65 00 20 00 43 00 6F 00 6E 00 74 00 72 00 6F 00 6C 00 6C 00 65 00 72 00 20 00 41 00 64 00 61 00 70 00 74 00 65 00 72 00 

[INTERFACE 0 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 00 00 61 
05 01 09 04 A1 01 09 01 A1 00 05 09 19 01 29 10 15 00 25 01 75 01 95 10 81 02 05 01 09 30 09 31 09 32 09 35 15 00 26 FF 00 75 08 95 04 81 02 05 01 09 33 09 34 95 02 81 02 05 01 09 39 15 00 25 07 35 00 46 3B 01 65 14 75 04 95 01 81 02 75 04 95 01 81 01 06 00 FF 09 01 75 08 95 02 91 02 C0 C0 

[INTERFACE 1 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 01 00 61 
05 01 09 04 A1 01 09 01 A1 00 05 09 19 01 29 10 15 00 25 01 75 01 95 10 81 02 05 01 09 30 09 31 09 32 09 35 15 00 26 FF 00 75 08 95 04 81 02 05 01 09 33 09 34 95 02 81 02 05 01 09 39 15 00 25 07 35 00 46 3B 01 65 14 75 04 95 01 81 02 75 04 95 01 81 01 06 00 FF 09 01 75 08 95 02 91 02 C0 C0 
`

Wii U/NS Mode:
`[USB DESCRIPTOR]
12 01 00 02 00 00 00 40 7E 05 37 03 00 01 01 02 03 01 09 02 29 00 01 01 00 E0 FA 09 04 00 00 02 03 00 00 00 09 21 10 01 00 01 22 D6 00 07 05 81 03 25 00 08 07 05 02 03 05 00 08 

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 12 
12 03 4E 00 69 00 6E 00 74 00 65 00 6E 00 64 00 6F 00 

[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 10 
10 03 57 00 55 00 50 00 2D 00 30 00 32 00 38 00 

[STRING 3 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 03 04 09 00 16 
16 03 31 00 35 00 2F 00 30 00 37 00 2F 00 32 00 30 00 31 00 34 00 

[INTERFACE 0 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 00 00 D6 
05 05 09 00 A1 01 85 11 19 00 2A FF 00 15 00 26 FF 00 75 08 95 05 91 00 C0 A1 01 85 21 19 00 2A FF 00 15 00 26 FF 00 75 08 95 25 81 00 C0 A1 01 85 12 19 00 2A FF 00 15 00 26 FF 00 75 08 95 01 91 00 C0 A1 01 85 22 19 00 2A FF 00 15 00 26 FF 00 75 08 95 19 81 00 C0 A1 01 85 13 19 00 2A FF 00 15 00 26 FF 00 75 08 95 01 91 00 C0 A1 01 85 23 19 00 2A FF 00 15 00 26 FF 00 75 08 95 02 81 00 C0 A1 01 85 14 19 00 2A FF 00 15 00 26 FF 00 75 08 95 01 91 00 C0 A1 01 85 24 19 00 2A FF 00 15 00 26 FF 00 75 08 95 02 81 00 C0 A1 01 85 15 19 00 2A FF 00 15 00 26 FF 00 75 08 95 01 91 00 C0 A1 01 85 25 19 00 2A FF 00 15 00 26 FF 00 75 08 95 02 81 00 C0 
`

J2Kbr, 2020-03-03:
Decoded Descriptors:

PC Mode:
```
[USB DESCRIPTOR]
12           bLength: 18
01           bDescriptorType: 1 - DEVICE
00 02        bcdUSB: 2.00
00           bDeviceClass: 0 (00)
00           bDeviceSubClass: 0 (00)
00           bDeviceProtocol: 0 (00)
40           bMaxPacketSize0: 64
79 00        idVendor: 0079
47 18        idProduct: 1847
00 01        bcdDevice: 1.00
01           iManufacturer: 1
02           iProduct: 2
00           iSerialNumber: 0
01           bNumConfigurations: 1

09           bLength: 9
02           bDescriptorType: 2 - CONFIGURATION
3B 00        wTotalLength: 59
02           bNumInterfaces: 2
01           bConfigurationValue: 1
00           iConfiguration: 0
C0           bmAttributes: 192 (Self Powered)
FA           MaxPower: 500mA

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
00             bInterfaceNumber: 0
00             bAlternateSetting: 0
01             bNumEndpoints: 1
03             bInterfaceClass: 3 (03)
00             bInterfaceSubClass: 0 (00)
00             bInterfaceProtocol: 0 (00)
00             iInterface: 0

09               bLength: 9
21               bDescriptorType: 33 - HID
10 01            bcdHID: 1.10
00               bCountryCode: 0
01               bNumDescriptors: 1
22               bDescriptorType: 34
61 00            wDescriptorLength: 97

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
81               bEndpointAddress: 1 IN
03               bmAttributes: 3 (Interrupt, None, Data)
40 00            wMaxPacketSize: 64
04               bInterval: 4

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
01             bInterfaceNumber: 1
00             bAlternateSetting: 0
01             bNumEndpoints: 1
03             bInterfaceClass: 3 (03)
00             bInterfaceSubClass: 0 (00)
00             bInterfaceProtocol: 0 (00)
00             iInterface: 0

09               bLength: 9
21               bDescriptorType: 33 - HID
10 01            bcdHID: 1.10
00               bCountryCode: 0
01               bNumDescriptors: 1
22               bDescriptorType: 34
[... 272 lines cut ...]
81 00          (MAIN)   INPUT              0x00000000 (2 fields x 8 bits) 0=Data 0=Array 0=Absolute 0=Ignored 0=Ignored 0=PrefState 0=NoNull 
C0           (MAIN)   END_COLLECTION     Application

A1 01        (MAIN)   COLLECTION         0x01 Application (Usage=0x0: Page=, Usage=, Type=)
85 15          (GLOBAL) REPORT_ID          0x15 (21) 
19 00          (LOCAL)  USAGE_MINIMUM      0x00050000 Undefined 
2A FF00        (LOCAL)  USAGE_MAXIMUM      0x000500FF  
15 00          (GLOBAL) LOGICAL_MINIMUM    0x00 (0)
26 FF00        (GLOBAL) LOGICAL_MAXIMUM    0x00FF (255)
75 08          (GLOBAL) REPORT_SIZE        0x08 (8) Number of bits per field
95 01          (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields 
91 00          (MAIN)   OUTPUT             0x00000000 (1 field x 8 bits) 0=Data 0=Array 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
C0           (MAIN)   END_COLLECTION     Application

A1 01        (MAIN)   COLLECTION         0x01 Application (Usage=0x0: Page=, Usage=, Type=)
85 25          (GLOBAL) REPORT_ID          0x25 (37) 
19 00          (LOCAL)  USAGE_MINIMUM      0x00050000 Undefined 
2A FF00        (LOCAL)  USAGE_MAXIMUM      0x000500FF  
15 00          (GLOBAL) LOGICAL_MINIMUM    0x00 (0)
26 FF00        (GLOBAL) LOGICAL_MAXIMUM    0x00FF (255)
75 08          (GLOBAL) REPORT_SIZE        0x08 (8) Number of bits per field
95 02          (GLOBAL) REPORT_COUNT       0x02 (2) Number of fields 
81 00          (MAIN)   INPUT              0x00000000 (2 fields x 8 bits) 0=Data 0=Array 0=Absolute 0=Ignored 0=Ignored 0=PrefState 0=NoNull 
C0           (MAIN)   END_COLLECTION     Application
```

J2Kbr, 2020-03-03:
In PC mode this controller should work with the Titan Two, have you already tested? You may need just perform button remap with an Input Translator or GPC script.

closed 2020-03-03 by J2Kbr

nelsontm, 2020-04-04:
A full press of ZL sets L to 100.0. A full press of ZR sets R to 100.0. L and R do nothing