J2Kbr/TitanTwo #219: Disable Memory Slot 0 access using Controller

oneswitch, 2019-07-30:
Arduino based controllers that are dependant upon T2 scripts to function "crash" when slot 0 is selected. 

It would be really useful to add a tick-box Device option to "Disable Memory Slot 0 access using controller". 

This would mean people using such controllers could continue to change slots freely, but never get trapped in slot 0 if unable to physically push the buttons on the actual Titan Two.

J2Kbr, 2020-03-03:
Hi Barrie. I spend some time trying find a good solution for this request and the problem is that the code that handles the memory slot change (previous/next) currently have no means to know the origin of the command (if from a button on the device, if from Gtuner, if from controller or if from a script or gamepack). And the code that generates the event (to change memory slot) has no information to anticipate if the next memory slot to load will be 0.

Of course, nothing is impossible and we could implement a solution for this, such as sharing information between above said modules. However, this solution would also require adding more conditional checking in the main loop of the device, and is here my main concern. I have all tasks in the main loop super hyper mega (you got the idea) optimized for speed and really don't want add any overhead here.

I hope you understand my point. Please let me know if there is anything I can do to help on the Arduino side, maybe we can find a solution there. ;)

closed 2020-03-03 by J2Kbr

oneswitch, 2020-03-04:
Thanks for looking into it. Understand the benefit of keeping speed. No
worries at all.

Barrie