J2Kbr/TitanTwo #223: Apple iOS POWER issue FIX needed for T2 connectivity (

oneswitch, 2019-08-11:
Please try your idea Jefferson, to "change the USB enumeration descriptors with lower power requirement [currently it's 500mA]. It seems this is all needed to make the Titan Two work on iOS."

This may make the Titan Two work with Apple iOS, especially so for the iOS 13 Beta, which might enable the T2 (with a Gamepad to Mouse script) to fully access an Apple iPhone/iPad.

Error messages here: [https://www.dropbox.com/home/work/specialeffect/1%20work%20in%20progress/iOS%20TITAN%20TWO/Power%20error%20message](Dropbox Pictures)

[image]

oneswitch, 2019-08-13:
Using a Lightning to USB 3 Camera Adapter, and connecting an iPad power block to that, the Titan Two connects to the iPad and works!

Using short leads, I can patch in the PC too.

[https://youtu.be/t3fp8bDMZOE](https://youtu.be/t3fp8bDMZOE) and [https://youtu.be/ZZudhvkeQcw](https://youtu.be/ZZudhvkeQcw).

Some issues remain.

1. On-screen keyboard hard to get to. Short-cut is: Activate a key (e.g. cursor key via d-pad using the gamepad to mouse script) - then click on the bottom right arrow for a few seconds. Wonder if it might be possible to offer a "disable keyboard" from HID multi protocol? Not ideal, so will be asking Apple for alternative easier access to the keyboard, but might stop the iOS from thinking a full keyboard is attached when it isn't.

2. Xbox Adaptive Controller. Seems really fussy on connecting. Sometimes it does, sometimes it doesn't. I have extra power connected to the USB adapter / T2 / XAC, and no combination seems to reliably make connection reliable. Also, bluetooth connectivity to the XAC doesn't seem to work full-stop.

3. Can't connect game controllers via USB on iOS13 Beta. Will ask Apple to do this. With Android we can hot swap slots between mouse protocol and Xbox 360/PS4 and it works well ([see here](https://youtu.be/DDxRO8eV0HU)). At the minute, on iOS13 it's only possible to have USB T2 connection for Mouse access. If I want Gamepad, I can only do it by physically disconnecting the T2, and connecting a PS4 Dualshock (for example) via Bluetooth. Then you get no T2 scripting benefits... Hoping Apple may add this functionality. Will ask.

4. I managed to get a power-pack running the T2 for the extra power it needed, as I imagined you could. Will look into this more. Useful for staying portable.

oneswitch, 2019-08-16:
[Blog post on iOS13 latest news with the Titan Two](https://switchgaming.blogspot.com/2019/08/gamepad-to-mouse-to-touch-ios-and.html)

**1. On Screen Keyboard:** In short, the on-screen keyboard issue has been solved (there's an option you can turn on so it always appears).

**2. Xbox Adaptive Controller:** I need to try this a bit more, but have posted a request for bluetooth connectivity to the Titan Two, which seems  (to me) to not work. Perhaps a slightly different standard. I could look into posting you one, Jefferson, if helpful?

**3. PS4/Xbox One USB output protocol to iOS:** This request has been added at Apple. Fingers crossed they'll be able to see the benefits and will implement it.

**4. Power:** It would be interesting to see if the Titan Two could function with iOS13 onwards without the need for extra power, but if bluetooth and power hungry attached controllers such as the XAC and Game Control Mixer wouldn't work, then maybe more important to just say extra power is needed as iPads don't provide it. I'm not sure.

J2Kbr, 2019-08-20:
On FW 1.04 I didn't change the power requirement in the USB descriptors (still 500mA), However, I set to self-powered (before was bus-powered). This should be enough to stop the iOS message to pop-up, and -if needed- the Titan Two can be powered directly, not requiring extra adapters.

closed 2019-08-20 by J2Kbr