J2Kbr/TitanTwo #225: Xbox Adaptive Controller Bluetooth Connection issue

oneswitch, 2019-08-16:
Would love to get the Microsoft Xbox Adaptive Controller possible to connect to the Titan Two via Bluetooth. Is there anything I can do to help with this?

J2Kbr, 2019-08-21:
Checking the firmware code I verified to have already added the VID:PID of the Xbox Adaptive Controller. Could you please give more info of what happens when trying connecting this controller over Bluetooth? Does it successfully pair with the Titan Two?

J2Kbr, 2019-08-21:
After further investigating, I found on the Internet that the Xbox Adaptive controller has a different "device name" from a Xbox One controllers when reporting itself over Bluetooth. On firmware 1.04, the correct device detection for the Xbox Adaptive Controller Bluetooth pairing was added. Hopefully with this change you will be able to use this controller wirelessly with the Titan Two.

closed 2019-08-21 by J2Kbr

oneswitch, 2019-09-21:
Many thanks. I'll try this out tomorrow.

oneswitch, 2019-09-22:
Works a treat, thanks so much!

[OneSwitch blog post](https://switchgaming.blogspot.com/2019/09/wireless-xac-to-ps4-using-titan-two.html)

I've passed this onto my colleagues at SE too as this will no doubt be another string to the Titan Twos bow for accessibility uses.

Great work as ever, Jefferson. :)

J2Kbr, 2019-09-22:
Awesome. Thank you for confirming it worked!! :) Very happy.