J2Kbr/TitanTwo #228: Wii-U Protocol fix needed

oneswitch, 2019-09-02:
Please see the thread at the [ConsoleTuner forums here](https://www.consoletuner.com/forum/viewtopic.php?f=27&t=13373&wpureload=1).

In short, Wii-U connection seems to work well for 1 and a half minutes, then goe bad (controls stop responding) for 15-20 seconds, in a repeating pattern.

oneswitch, 2019-09-13:
Thanks for this Jefferson. I've updated GTuner (no update for the firmware available). I'm finding connecting to the Wii-U much easier. Not sure if I've just found how to do it reliably.

0. Titan Two programmed to Wii-U protocol + Left Hand Translator in slot 1 + slot 1 selected. Wii-U up to date Gamepad and Wii-U remote with good batteries pre-synched.

1. Everything OFF.
2. Connect Wii U Pro controller > T2 > T2 output > Wii-U USB
3. Wii Remote ON (press A)
4. Pair Wii U Pro Controller to T2 (T2 UP for 4 seconds, then Pro sync button). Wait 10 seconds.
5. Pair T2 to Wii U console (T2 UP for 4 seconds, then Wii-U console sync button). Wait 20 seconds.
6. Wii Remote (press B to go back)
7. Wii Pro controller (HOME > HOME > "Controller Settings" > A > "Wii Remote/Other Controller Settings" > A > Change Order > A > HOME > B > B > B > to Wii U menu.... Disconnect Wii Pro Controller from T2 if you wish.

That works great in the main. However, I'm still getting the drop out every 1 and a half minutes for 15-20 seconds. Not only in Mario Kart 8 but anything.

Video here: [https://youtu.be/9IO2_gXfg8o](https://youtu.be/9IO2_gXfg8o)

Wonder what it might be.

J2Kbr, 2019-09-13:
Thanks, the firmware with this udpade was not yet made available, I will try get it out today or -at the most- during the weekend.

oneswitch, 2019-09-18:
Seems fixed, if using a wired controller.

No Wii access (unless hacked you said).
Wii-U access requires blue-tooth kit, up to date firmware, and forced Wii U protocol. Very stable.
Used Xbox 360 wired controller successfully.

Many thanks!