J2Kbr/TitanTwo #232: beta fw 1.5.1.2 - did the mouse_set handling change ?

Scachi, 2019-09-09:
I have an older usb hid script that uses this combo for rapid fire for Output Protocol "USB Multi Interface HID" that isn working anymore.
Not sure if this only happens because of the beta as I have both my T2 updated to beta for running other tests and can't downgrade back to stable.
Also I am not sure if I have changed my setup and it isn't working anymore because of using the mouse/keyboard differently. I haven't run the usb-hid script and setup for a long time.

Here is the old combo:
```
combo cRapid {
	mouse_set(MBUTTON_1,1);           // fire
	wait(0); wait(RapidHold);         // time the fire button is pressed
	mouse_set(MBUTTON_1,0);           // release
	wait(0); wait(RapidWait);         // time the fire button is released
}   
```

Now I have to use it like this:
```
combo cRapidFire {
  wait(RapidHold);	// time the fire button is pressed
  mouse_set(MBUTTON_1,0);  	// release
  wait(RapidWait);	// time the fire button is released
}
```

Scachi, 2019-09-10:
The whole set_mouse stuff seems to be a bit hard to use.
When I want to set a mouse button pressed or released only I end up with unwanted ghost mouse movement in the direction of the last input just by using mouse_set(MBUTTON_1,1);

I have tried a lot of different approaches and it looks like I have got it working reliable now.
Not sure if I am missing something important on how the mouse_set USB-HID stuff is supposed to work or if there is the chance to improve this mouse_set/USB-HID behaviour to make it more userfriendly.

The current version v1.05 of the script attached.
It is working reliable with the beta firmware only 1.5.1.2.
[usb_hid_antirecoil_v1.05.zip](https://github.com/J2Kbr/TitanTwo/files/3596520/usb_hid_antirecoil_v1.05.zip)

J2Kbr, 2019-09-10:
What you are describing is not related with the changes made recently, not sure how your script worked as expected before.

The issue is due the combination of `mouse_passthru()` with `mouse_set()`. The first function will overwrite the output buffer with the latest data received from the connected mouse, this operation will also overwrite the value previously set by `mouse_set()` (in the combo with `wait(0)` ).

To mitigate this situation, you can check if a new data report was received from the connected mouse before call `mouse_passthru()`, example:

```
#include <mouse.gph>
 
init {
    mousemapping();
}
 
main {
    if(mouse_status(MREPORT_UPDATED)) {
        mouse_passthru();
    }
}
```

closed 2019-09-10 by J2Kbr

Scachi, 2019-09-10:
Thank you very much. That explains a lot.