J2Kbr/TitanTwo #233: Titan Two Xbox One Wireless Receiver Disconnecting

Avoidies, 2019-09-10:
Hello, I'm having a issue with the titan two on the Xbox one. The problem that's happening is after I sync my controller to the titan two. Pressing the guide button a few times it disconnects from the titan two and it holds the guide button down. My controller will stay on and on the titan two it will say it's authorized while the scrips are still loaded but my controller will keep holding the guide button if I try to press on any button or move the analog sticks.I have my SD card already configured and my wireless adapter connected to the titan two. I have the latest update on my titan two as well(1.5.1.2).

Scachi, 2019-09-10:
Have you tried the latest stable version:  1.4.3.1 ?
You can set the beta entry to false, put your T2 into P mode by holding down both buttons  and start gtuner IV, then Tools->Check for Updates to downgrade to stable again to test if it works for you.

Just linking this for J2kbr. An issue like this was reported and fixed before: 
https://github.com/J2Kbr/TitanTwo/issues/204

Avoidies, 2019-09-10:
Yes it was still disconnecting. I had just updated my firmware from that version and it disconnected then and now.

J2Kbr, 2019-09-13:
The revision 6 of FW 1.05B will bring improvements to the Xbox One Wireless Receiver support. Is expected the new revision to be available during this weekend. After updating your Titan Two please check if it solved the disconnection issue. Thanks.

closed 2019-09-13 by J2Kbr

Avoidies, 2019-09-14:
Okay thank you so much, has been an ongoing problem since getting the Titan Two.