J2Kbr/TitanTwo #240: T2 on XB360 console seems not to be working at the moment

Scachi, 2019-09-27:
There seems to be an issue with the T2 working on XB360 console at the moment.
With wired xb360 controller there seems to be no auth working correcltly and/or no control üassing through to the connected cosole while the Device Monitor show Input and Output.

Videos by user  pauline1108:
https://youtu.be/W0y_L8Lkcis
https://youtu.be/9lyW4j7LWxg
Screenshot of the device monitor: 
[image]

Full forum thread: https://www.consoletuner.com/forum/viewtopic.php?f=27&t=13647

oneswitch, 2019-09-27:
Additionally, I've noticed a drop out in connection (Xbox 360 protocol with wired Xbox 360 controller) on Windows 10 and Xbox 360. I've tried 3 different official Xbox 360 wired controllers, all with the same issue on both machines.

On PC I test by running the joy.cpl utility, or running a game in Steam. Drop outs seem to occur slightly randomly but quite regularly (within a few minutes). Sometimes you'll see repeated drop outs.

On Xbox 360 with a game "Please reconnect controller" appears quite regularly too.

Similarly on a Raspberry Pi/RetroPie, recent T2's are dropping out, but my much older ones that have not been updated, are not. Sorry not very specific beyond that. Anything I can do more to help, just ask.

Barrie

J2Kbr, 2019-11-20:
Retested with several XB360 controller I have available, all worked. I assume the issue is related with the user's third-party controller (from Inno GamePad).

closed 2019-11-20 by J2Kbr