J2Kbr/TitanTwo #243: Add direct horizontal scroll for mouse passthrough

AeliusSaionji, 2019-11-08:
I am aware that shift+mwheel will scroll horizontally, but unfortunately this means I cannot scroll vertically and horizontally at the same time. I am trying to convert mouse movement into 2D scrolling (for use with a trackball mouse, for panning pictures, or in Audacity, etc).

Without the ability to send both scroll directions to the OS at the same time, the implementation is stuttery.

Is this possible?
Thanks.

AeliusSaionji, 2020-01-22:
For clarity:
I have written a script which attempts to convert mouse movement into 2D scrolling. Horizontal movement sends `shift+scroll` , vertical movement sends `scroll`.
When I move the mouse diagonally, as you might expect it does not work well, and having `shift` constantly sent to the PC is also rather intrusive in some contexts.

J2Kbr, 2020-03-03:
Feature added to Gtuner IV 1.07 and Titan Two firmware 1.07 (will be available during the beta period).

Works similarly as MOUSE_WHEEL, example:

```
#include <mouse.gph>

main {
    mouse_set(MOUSE_ACPAN, 1);
}
```

closed 2020-03-03 by J2Kbr