J2Kbr/TitanTwo #247: Unique Interactions with moonlight controller emulation

Seishirou101, 2019-11-14:
When trying to use moonlight android application in kmg capture, the inputs go semi crazy when detecting. As seen by this video https://streamable.com/ty0fv. The only application that exhibits this behavior is gtuner, all other programs I've tested have behaved how html5 gamepad tester is shown. 
Parsec, a similar app does not exhibit this behavior with their controller emulator. 

The controller is an 8bitdo sn30 pro + as well

J2Kbr, 2020-03-03:
Thank you for the video, it helped fix a bug (see in the device monitor, there are 2 OUT connections, which is wrong). I believe was this bug that was causing the reported issue. The fix will be available in the Titan Two firmware 1.07.

Seishirou101, 2020-03-27:
After updating to 1.7.3.0 firmware, (I'm assuming 1.07 is a typo) the issue still exists, the behavior is exactly the same as the video but the USB C is no longer present, as stated by the bug that was fixed with firmware 1.7.0. Parsec has also changed how their controller emulation is done and mirrors moonlight's controller emulation behavior as seen by the past video. See attached screenshot https://puu.sh/FpNxO/e56b4b075d.png

J2Kbr, 2020-05-08:
I am closing this as #262 is related to the same issue.

closed 2020-05-08 by J2Kbr