J2Kbr/TitanTwo #248: Wii compatibility

DuckCurious, 2019-11-21:
As mentioned in a thread in the support forum, the Titan Two cannot connect with an original Wii currently, even though it is stated explicitly in the documentation that it can. Would love to see this implemented, or at least update the documentation to remove the claims that this device does support the OG Wii, as it is currently very misleading.

MaxRat08, 2020-03-21:
I am personally extremely interested in support for the original Wii, the ability to play Wii games with a "normal" controller was the initial reason I purchased a Titan Two.

J2Kbr, 2020-05-12:
After 4 days exclusively working on the Wii support, it was verified emulate a Wiimote controller is not viable, specially due the accelerometer and IR sensors. The Titan Two could emulate a Wiimote with Classic controller attachment. However, limitations with Wii UI, the realization this option would work with just few games and the amount of flash memory such support would take from the Titan Two, it was concluded the efforts would not worth the benefits.

The Titan Two documentation pages have been updated to remove references to Wii console support, as well as corresponding references on Gtuner IV/Air software.

**Side note:** _With jail-broken Wii consoles, the Titan Two can be used as, in this case, the Wii U Pro controller is supported by the console._ 

Thank you for your understanding.

closed 2020-05-12 by J2Kbr