J2Kbr/TitanTwo #250: Stinky footboard descriptors

NeoTechni, 2019-12-29:
It's basically a foot-operated DPAD

[USB DESCRIPTOR]
12 01 10 01 00 00 00 08 E0 27 01 01 01 00 01 02 03 01 09 02 3B 00 02 01 00 C0 32 09 04 00 00 01 03 00 00 00 09 21 11 01 00 01 22 3F 00 07 05 81 03 08 00 01 09 04 01 00 01 03 00 00 00 09 21 11 01 00 01 22 20 00 07 05 82 03 40 00 01 

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 18 
18 03 53 00 74 00 65 00 6C 00 75 00 6C 00 75 00 00 00 1C 03 53 00 74 00 

[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 1C 
1C 03 53 00 74 00 69 00 6E 00 6B 00 79 00 20 00 42 00 6F 00 61 00 72 00 64 00 00 00 

[STRING 3 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 03 04 09 00 1A 
1A 03 30 00 30 00 30 00 30 00 30 00 30 00 30 00 30 00 30 00 30 00 30 00 30 00 

[INTERFACE 0 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 00 00 3F 
05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01 75 01 95 08 81 02 95 01 75 08 81 01 05 08 19 01 29 05 95 05 75 01 91 02 95 01 75 03 91 01 15 00 25 65 05 07 19 00 29 65 95 06 75 08 81 00 C0 

[INTERFACE 1 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 01 00 20 
06 00 FF 09 01 A1 01 09 02 15 00 25 FF 75 08 95 40 81 02 09 03 15 00 25 FF 75 08 95 40 91 02 C0 

[USB SNIFF]
00 00 00 01 00 00 01 00 00 05 00 10 00 00 00 00 
00 00 00 02 00 00 02 01 00 05 00 03 00 00 00 00 
10 
00 00 00 01 00 00 01 00 80 06 01 00 00 00 00 08 
00 00 00 02 00 00 02 01 80 06 01 00 00 00 00 08 
12 01 10 01 00 00 00 08 
00 00 00 01 00 00 01 00 80 06 01 00 00 00 00 12 
00 00 00 02 00 00 02 01 80 06 01 00 00 00 00 12 
12 01 10 01 00 00 00 08 E0 27 01 01 01 00 01 02 03 01 
00 00 00 01 00 00 01 00 80 06 03 00 00 00 00 02 
00 00 00 02 00 00 02 01 80 06 03 00 00 00 00 02 
04 03 
00 00 00 01 00 00 01 00 80 06 03 00 00 00 00 04 
00 00 00 02 00 00 02 01 80 06 03 00 00 00 00 04 
04 03 09 04 
00 00 00 01 00 00 01 00 80 06 03 03 04 09 00 02 
00 00 00 02 00 00 02 01 80 06 03 03 04 09 00 02 
1A 03 
00 00 00 01 00 00 01 00 80 06 03 03 04 09 00 1A 
00 00 00 02 00 00 02 01 80 06 03 03 04 09 00 1A 
1A 03 30 00 30 00 30 00 30 00 30 00 30 00 30 00 30 00 30 00 30 00 30 00 30 00 
00 00 00 01 00 00 01 00 80 06 03 01 04 09 00 02 
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 02 
18 03 
00 00 00 01 00 00 01 00 80 06 03 01 04 09 00 18 
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 18 
18 03 53 00 74 00 65 00 6C 00 75 00 6C 00 75 00 00 00 1C 03 53 00 74 00 
00 00 00 01 00 00 01 00 80 06 03 02 04 09 00 02 
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 02 
1C 03 
00 00 00 01 00 00 01 00 80 06 03 02 04 09 00 1C 
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 1C 
1C 03 53 00 74 00 69 00 6E 00 6B 00 79 00 20 00 42 00 6F 00 61 00 72 00 64 00 00 00 
00 00 00 01 00 00 01 00 80 06 02 00 00 00 00 09 
00 00 00 02 00 00 02 01 80 06 02 00 00 00 00 09 
09 02 3B 00 02 01 00 C0 32 
00 00 00 01 00 00 01 00 80 06 02 00 00 00 00 3B 
00 00 00 02 00 00 02 01 80 06 02 00 00 00 00 3B 
09 02 3B 00 02 01 00 C0 32 09 04 00 00 01 03 00 00 00 09 21 11 01 00 01 22 3F 00 07 05 81 03 08 00 01 09 04 01 00 01 03 00 00 00 09 21 11 01 00 01 22 20 00 07 05 82 03 40 00 01 
00 00 00 01 00 00 01 00 80 00 00 00 00 00 00 02 
00 00 00 02 00 00 02 01 80 00 00 00 00 00 00 02 
00 00 
00 00 00 01 00 00 01 00 00 09 00 01 00 00 00 00 
00 00 00 02 00 00 02 01 00 09 00 01 00 00 00 00 
00 00 00 01 00 00 01 00 81 06 22 00 00 00 00 3F 
00 00 00 02 00 00 02 01 81 06 22 00 00 00 00 3F 
05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01 75 01 95 08 81 02 95 01 75 08 81 01 05 08 19 01 29 05 95 05 75 01 91 02 95 01 75 03 91 01 15 00 25 65 05 07 19 00 29 65 95 06 75 08 81 00 C0 
00 00 00 01 00 00 01 00 21 0A 00 00 00 00 00 00 
00 00 00 02 00 00 02 01 21 0A 00 00 00 00 00 00 
00 00 00 01 00 00 01 00 81 06 22 00 00 01 00 20 
00 00 00 02 00 00 02 01 81 06 22 00 00 01 00 20 
06 00 FF 09 01 A1 01 09 02 15 00 25 FF 75 08 95 40 81 02 09 03 15 00 25 FF 75 08 95 40 91 02 C0 
00 00 00 01 00 00 01 00 21 0A 00 00 00 01 00 00 
00 00 00 02 00 00 02 01 21 0A 00 00 00 01 00 00 
00 00 00 02 00 01 
30 00 80 00 00 00 00 00 
00 00 00 02 00 02 
30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 02 00 01 
00 00 80 00 00 00 00 00

AeliusSaionji, 2020-01-22:
I have this- does it really not work as a keyboard? You should be able to configure the device to be detected as a normal keyboard via the provided software.

NeoTechni, 2020-01-22:
I see no such option, nor an option to detect keyboards plugged into the Titan

NeoTechni, 2020-01-22:
Ah. So when you remap something in the stinky program, it stores it on the device. I expected it to work as a controller. Thank you.

closed 2020-03-03 by J2Kbr