J2Kbr/TitanTwo #253: Xbox Adaptive Controller via Bluetooth - VIEW and MENU not recognised

oneswitch, 2020-01-11:
VIEW and MENU are a problem via bluetooth.

They are recognised (via an Xbox Adaptive Controller) if connected via USB-A or USB-B. They are not via bluetooth. The Xbox Home button and everything else does seem to be fine.

oneswitch, 2020-01-13:
I'm testing with a prototype XAC with firmware at 1.6.3.0. I'll try to get hold of a production run unit to see if the firmware goes higher than that, perhaps explaining the problem I'm seeing.

Anyone else having this problem would do well to update the firmware on their XACs via the Xbox One settings > accessories page, or via the Microsoft Store Xbox accessories app.

oneswitch, 2020-01-18:
Small update:

LS = nothing
LB = VIEW
RB = MENU
RS = nothing
X = Y
Y = LB

Hopefully a firmware update to the T2 will address this.

J2Kbr, 2020-01-18:
Fixed on FW 2.06 Revision 1 (to be release soon).

closed 2020-01-18 by J2Kbr

oneswitch, 2020-01-20:
So close! Everything seems good, apart from the home button. On a Xbox Adaptive Controller the home-button is not working at all. As I understand, it has a different assignation to a standard Xbox One controller (although on the surface doing exactly the same thing).

J2Kbr, 2020-01-20:
Thank you for the feedback. I have already fixed the Xbox button, the new FW update will be available soon.