J2Kbr/TitanTwo #254: PSVR 3DRudder USB Descriptors

AeliusSaionji, 2020-01-22:
[USB DESCRIPTOR]
12 01 00 02 00 00 00 08 FA 2D 02 00 23 24 01 02 00 01 09 02 29 00 01 01 00 80 32 09 04 00 00 02 03 00 00 00 09 21 11 01 00 01 22 1D 00 07 05 81 03 40 00 01 07 05 01 03 40 00 01 

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 1C 
1C 03 33 00 44 00 52 00 75 00 64 00 64 00 65 00 72 00 20 00 49 00 6E 00 63 00 2E 00 

[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 12 
12 03 33 00 44 00 52 00 55 00 44 00 44 00 45 00 52 00 

[INTERFACE 0 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 00 00 1D 
06 00 FF 09 01 A1 01 85 01 15 00 26 FF 00 75 08 95 3F 09 01 81 02 95 40 09 01 91 00 C0

AeliusSaionji, 2020-01-22:
The PSVR variant of the 3DRudder. Unlike the PC version, this has no generic capabilities that a PC can recognize. Curious to see if the Titan can read it.

J2Kbr, 2020-03-03:
Indeed, this device has only a vendor specific interface, as can be verified by the following decoded descriptors:
```
[USB DESCRIPTOR]
12           bLength: 18
01           bDescriptorType: 1 - DEVICE
00 02        bcdUSB: 2.00
00           bDeviceClass: 0 (00)
00           bDeviceSubClass: 0 (00)
00           bDeviceProtocol: 0 (00)
08           bMaxPacketSize0: 8
FA 2D        idVendor: 2DFA
02 00        idProduct: 0002
23 24        bcdDevice: 24.23
01           iManufacturer: 1
02           iProduct: 2
00           iSerialNumber: 0
01           bNumConfigurations: 1

09           bLength: 9
02           bDescriptorType: 2 - CONFIGURATION
29 00        wTotalLength: 41
01           bNumInterfaces: 1
01           bConfigurationValue: 1
00           iConfiguration: 0
80           bmAttributes: 128 (Bus Powered)
32           MaxPower: 100mA

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
00             bInterfaceNumber: 0
00             bAlternateSetting: 0
02             bNumEndpoints: 2
03             bInterfaceClass: 3 (03)
00             bInterfaceSubClass: 0 (00)
00             bInterfaceProtocol: 0 (00)
00             iInterface: 0

09               bLength: 9
21               bDescriptorType: 33 - HID
11 01            bcdHID: 1.11
00               bCountryCode: 0
01               bNumDescriptors: 1
22               bDescriptorType: 34
1D 00            wDescriptorLength: 29

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
81               bEndpointAddress: 1 IN
03               bmAttributes: 3 (Interrupt, None, Data)
40 00            wMaxPacketSize: 64
01               bInterval: 1

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
01               bEndpointAddress: 1 OUT
03               bmAttributes: 3 (Interrupt, None, Data)
40 00            wMaxPacketSize: 64
01               bInterval: 1

[STRING 1 DESCRIPTOR]
3DRudder Inc.

[STRING 2 DESCRIPTOR]
3DRUDDER

[INTERFACE 0 HID DESCRIPTOR]
06 00FF      (GLOBAL) USAGE_PAGE         0xFF00 Vendor-defined 
09 01        (LOCAL)  USAGE              0xFF000001  
A1 01        (MAIN)   COLLECTION         0x01 Application (Usage=0xFF000001: Page=Vendor-defined, Usage=, Type=)
85 01          (GLOBAL) REPORT_ID          0x01 (1) 
15 00          (GLOBAL) LOGICAL_MINIMUM    0x00 (0)
26 FF00        (GLOBAL) LOGICAL_MAXIMUM    0x00FF (255) 
75 08          (GLOBAL) REPORT_SIZE        0x08 (8) Number of bits per field 
95 3F          (GLOBAL) REPORT_COUNT       0x3F (63) Number of fields 
09 01          (LOCAL)  USAGE              0xFF000001  
81 02          (MAIN)   INPUT              0x00000002 (63 fields x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
95 40          (GLOBAL) REPORT_COUNT       0x40 (64) Number of fields 
09 01          (LOCAL)  USAGE              0xFF000001  
91 00          (MAIN)   OUTPUT             0x00000000 (64 fields x 8 bits) 0=Data 0=Array 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
C0           (MAIN)   END_COLLECTION     Application
```
With only this information there is not much I can do to support this device. I would need more details about the protocol and data structure in order to read the inputs and map to the respective GPC designators.

If eventually you find more details, please comment and I reopen this request.

Thanks.

closed 2020-03-03 by J2Kbr