J2Kbr/TitanTwo #255: Microsoft steering wheel - add to recognized devices

tom10mount, 2020-01-26:
HI, 
I got a Microsoft steering wheel and it seems not to be one of the compatible devices so my Titan Two is not recognizing it. I ran the Fetch descriptors, see below. Is it possible to get this unit added to the compatible devices?

[USB DESCRIPTOR]
12 01 00 01 00 00 00 08 5E 04 34 00 01 01 01 02 00 01 09 02 29 00 01 01 00 A0 32 09 04 00 00 02 03 00 00 00 09 21 00 01 00 01 22 10 05 07 05 81 03 40 00 01 07 05 02 03 10 00 04 

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 14 
14 03 4D 00 69 00 63 00 72 00 6F 00 73 00 6F 00 66 00 74 00 

[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 4C 
4C 03 53 00 69 00 64 00 65 00 57 00 69 00 6E 00 64 00 65 00 72 00 20 00 46 00 6F 00 72 00 63 00 65 00 20 00 46 00 65 00 65 00 64 00 62 00 61 00 63 00 6B 00 20 00 57 00 68 00 65 00 65 00 6C 00 20 00 28 00 55 00 53 00 42 00 29 00 

[INTERFACE 0 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 00 05 10 
05 01 09 04 A1 01 85 01 09 30 16 00 FE 26 FF 01 35 00 46 FF 03 75 0A 95 01 81 02 75 06 81 03 A1 00 05 01 09 31 15 00 25 3F 35 00 45 3F 75 06 95 01 81 02 75 02 81 03 09 35 75 06 81 02 75 02 81 03 C0 05 09 15 00 19 01 29 08 25 01 35 00 45 01 75 01 95 08 81 02 06 01 FF 09 49 75 01 95 01 81 02 75 07 81 03 05 0F 09 92 A1 02 85 02 09 9F 09 A0 09 A4 09 A5 09 A6 15 00 25 01 35 00 45 01 75 01 95 05 81 02 95 03 81 03 09 94 15 00 25 01 35 00 45 01 75 01 95 01 81 02 09 22 15 01 25 28 35 01 45 28 75 07 95 01 81 02 C0 09 21 A1 02 85 01 09 22 15 01 25 28 35 01 45 28 75 08 95 01 91 02 09 25 A1 02 09 26 09 27 09 30 09 31 09 32 09 33 09 34 09 40 09 41 09 42 09 43 09 28 25 0C 15 01 35 01 45 0C 75 08 95 01 91 00 C0 09 50 09 54 09 51 15 00 26 FF 7F 35 00 46 FF 7F 66 03 10 55 FD 75 10 95 03 91 02 55 00 66 00 00 09 52 15 00 26 FF 00 35 00 46 10 27 75 08 95 01 91 02 09 53 15 01 25 08 35 01 45 08 75 08 95 01 91 02 09 55 A1 02 05 01 09 30 09 31 15 00 25 01 75 01 95 02 91 02 C0 05 0F 09 56 95 01 91 02 95 05 91 03 09 57 A1 02 0B 01 00 0A 00 0B 02 00 0A 00 66 14 00 55 FE 15 00 26 FF 00 35 00 47 A0 8C 00 00 66 00 00 75 08 95 02 91 02 55 00 66 00 00 C0 05 0F 09 A7 66 03 10 55 FD 15 00 26 FF 7F 35 00 46 FF 7F 75 10 95 01 91 02 66 00 00 55 00 C0 05 0F 09 5A A1 02 85 02 09 22 15 01 25 28 35 01 45 28 75 08 95 01 91 02 09 5B 09 5D 15 00 26 FF 00 35 00 46 10 27 95 02 91 02 09 5C 09 5E 66 03 10 55 FD 26 FF 7F 46 FF 7F 75 10 91 02 45 00 66 00 00 55 00 C0 09 5F A1 02 85 03 09 22 15 01 25 28 35 01 45 28 75 08 95 01 91 02 09 23 15 00 25 01 35 00 45 01 75 04 95 01 91 02 09 58 A1 02 0B 01 00 0A 00 0B 02 00 0A 00 75 02 95 02 91 02 C0 15 80 25 7F 36 F0 D8 46 10 27 09 60 75 08 95 01 91 02 36 F0 D8 46 10 27 09 61 09 62 95 02 91 02 15 00 26 FF 00 35 00 46 10 27 09 63 09 64 75 08 95 02 91 02 09 65 46 10 27 95 01 91 02 C0 09 6E A1 02 85 04 09 22 15 01 25 28 35 01 45 28 75 08 95 01 91 02 09 70 15 00 26 FF 00 35 00 46 10 27 75 08 95 01 91 02 09 6F 15 80 25 7F 36 F0 D8 46 10 27 95 01 91 02 09 71 66 14 00 55 FE 15 00 26 FF 00 35 00 47 A0 8C 00 00 91 02 09 72 26 FF 7F 46 FF 7F 66 03 10 55 FD 75 10 95 01 91 02 66 00 00 55 00 C0 09 73 A1 02 85 05 09 22 15 01 25 28 35 01 45 28 75 08 95 01 91 02 09 70 16 01 FF 26 FF 00 36 F0 D8 46 10 27 75 10 95 01 91 02 C0 09 74 A1 02 85 06 09 22 15 01 25 28 35 01 45 28 75 08 95 01 91 02 09 75 09 76 15 80 25 7F 36 F0 D8 46 10 27 75 08 95 02 91 02 C0 09 68 A1 02 85 07 09 22 15 01 25 28 35 01 45 28 75 08 95 01 91 02 09 6C 15 00 26 10 27 35 00 46 10 27 75 10 95 01 91 02 09 69 15 81 25 7F 35 00 46 FF 00 75 08 95 0C 92 02 01 C0 09 66 A1 02 85 08 05 01 09 30 09 31 15 81 25 7F 35 00 46 FF 00 75 08 95 02 91 02 C0 05 0F 09 77 A1 02 85 0A 09 22 15 01 25 28 35 01 45 28 75 08 95 01 91 02 09 78 A1 02 09 79 09 7A 09 7B 15 01 25 03 75 08 95 01 91 00 C0 09 7C 15 00 26 FF 00 35 00 46 FF 00 91 02 C0 09 90 A1 02 85 0B 09 22 25 28 15 01 35 01 45 28 75 08 95 01 91 02 C0 09 96 A1 02 85 0C 09 97 09 98 09 99 09 9A 09 9B 09 9C 15 01 25 06 75 08 95 01 91 00 C0 09 7D A1 02 85 0D 09 7E 15 00 26 FF 00 35 00 46 10 27 75 08 95 01 91 02 C0 09 6B A1 02 85 0E 09 22 15 01 25 28 35 01 45 28 75 08 95 01 91 02 09 6D 15 00 26 FF 00 35 00 46 FF 00 75 08 95 01 91 02 09 51 66 03 10 55 FD 15 00 26 FF 7F 35 00 46 FF 7F 75 10 95 01 91 02 55 00 66 00 00 C0 09 AB A1 02 85 01 09 25 A1 02 09 26 09 27 09 30 09 31 09 32 09 33 09 34 09 40 09 41 09 42 09 43 09 28 25 0C 15 01 35 01 45 0C 75 08 95 01 B1 00 C0 05 01 09 3B 15 00 26 FF 01 35 00 46 FF 01 75 0A 95 01 B1 02 75 06 B1 01 C0 05 0F 09 89 A1 02 85 02 09 22 25 28 15 01 35 01 45 28 75 08 95 01 B1 02 09 8B A1 02 09 8C 09 8D 09 8E 25 03 15 01 35 01 45 03 75 08 95 01 B1 00 C0 09 AC 15 00 27 FF FF 00 00 35 00 47 FF FF 00 00 75 10 95 01 B1 00 C0 09 7F A1 02 85 03 09 80 75 10 95 01 15 00 35 00 27 FF FF 00 00 47 FF FF 00 00 B1 02 09 83 26 FF 00 46 FF 00 75 08 95 01 B1 02 09 A9 09 AA 75 01 95 02 15 00 25 01 35 00 45 01 B1 02 75 06 95 01 B1 03 C0 C0

J2Kbr, 2020-03-03:
Below the descriptors decoded for future reference.

After a quick check, it seems the issue is the same as reported here: https://github.com/J2Kbr/TitanTwo/issues/257

The 1.07 update will be release this week. Please check if this wheel works after updating your device.

Thank you.

```
[USB DESCRIPTOR]
12           bLength: 18
01           bDescriptorType: 1 - DEVICE
00 01        bcdUSB: 1.00
00           bDeviceClass: 0 (00)
00           bDeviceSubClass: 0 (00)
00           bDeviceProtocol: 0 (00)
08           bMaxPacketSize0: 8
5E 04        idVendor: 045E
34 00        idProduct: 0034
01 01        bcdDevice: 1.01
01           iManufacturer: 1
02           iProduct: 2
00           iSerialNumber: 0
01           bNumConfigurations: 1

09           bLength: 9
02           bDescriptorType: 2 - CONFIGURATION
29 00        wTotalLength: 41
01           bNumInterfaces: 1
01           bConfigurationValue: 1
00           iConfiguration: 0
A0           bmAttributes: 160 (Bus Powered, Remote Wakeup)
32           MaxPower: 100mA

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
00             bInterfaceNumber: 0
00             bAlternateSetting: 0
02             bNumEndpoints: 2
03             bInterfaceClass: 3 (03)
00             bInterfaceSubClass: 0 (00)
00             bInterfaceProtocol: 0 (00)
00             iInterface: 0

09               bLength: 9
21               bDescriptorType: 33 - HID
00 01            bcdHID: 1.00
00               bCountryCode: 0
01               bNumDescriptors: 1
22               bDescriptorType: 34
10 05            wDescriptorLength: 1296

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
81               bEndpointAddress: 1 IN
03               bmAttributes: 3 (Interrupt, None, Data)
40 00            wMaxPacketSize: 64
01               bInterval: 1

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
02               bEndpointAddress: 2 OUT
03               bmAttributes: 3 (Interrupt, None, Data)
10 00            wMaxPacketSize: 16
04               bInterval: 4

[STRING 1 DESCRIPTOR]
Microsoft

[STRING 2 DESCRIPTOR]
[... 592 lines cut ...]
35 00            (GLOBAL) PHYSICAL_MINIMUM   0x00 (0)
27 FFFF0000      (GLOBAL) LOGICAL_MAXIMUM    0x0000FFFF (65535)
47 FFFF0000      (GLOBAL) PHYSICAL_MAXIMUM   0x0000FFFF (65535)
B1 02            (MAIN)   FEATURE            0x00000002 (1 field x 16 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
09 83            (LOCAL)  USAGE              0x000F0083 Simultaneous Effects Max (SV=Static Value) 
26 FF00          (GLOBAL) LOGICAL_MAXIMUM    0x00FF (255) 
46 FF00          (GLOBAL) PHYSICAL_MAXIMUM   0x00FF (255) 
75 08            (GLOBAL) REPORT_SIZE        0x08 (8) Number of bits per field 
95 01            (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields
B1 02            (MAIN)   FEATURE            0x00000002 (1 field x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
09 A9            (LOCAL)  USAGE              0x000F00A9 Device Managed Pool (SF=Static Flag) 
09 AA            (LOCAL)  USAGE              0x000F00AA Shared Parameter Blocks (SF=Static Flag) 
75 01            (GLOBAL) REPORT_SIZE        0x01 (1) Number of bits per field 
95 02            (GLOBAL) REPORT_COUNT       0x02 (2) Number of fields 
15 00            (GLOBAL) LOGICAL_MINIMUM    0x00 (0)
25 01            (GLOBAL) LOGICAL_MAXIMUM    0x01 (1) 
35 00            (GLOBAL) PHYSICAL_MINIMUM   0x00 (0)
45 01            (GLOBAL) PHYSICAL_MAXIMUM   0x01 (1) 
B1 02            (MAIN)   FEATURE            0x00000002 (2 fields x 1 bit) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
75 06            (GLOBAL) REPORT_SIZE        0x06 (6) Number of bits per field 
95 01            (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields 
B1 03            (MAIN)   FEATURE            0x00000003 (1 field x 6 bits) 1=Constant 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
C0             (MAIN)   END_COLLECTION     Logical
C0           (MAIN)   END_COLLECTION     Application
```

tom10mount, 2020-03-12:
Hi, I don’t see the 1.07 update yet. Any update on the release date?
Thanks,

J2Kbr, 2020-03-12:
Currently the 1.07 is available for beta testing, you can download from here:

https://www.consoletuner.com/forum/viewtopic.php?f=2&t=14717
https://www.consoletuner.com/forum/viewtopic.php?f=2&t=14879

tom10mount, 2020-03-12:
Thank you so much, I will give it a try.
Best regards,
Tom

https://www.consoletuner.com/forum/viewtopic.php?f=2&t=14717
https://www.consoletuner.com/forum/viewtopic.php?f=2&t=14879

tom10mount, 2020-03-12:
Hello,
I have got my T2 updated, thanks. But to be honest I’m not sure wat to do next to set up an input translator so my steering wheel will work with my x-box one.  I you have a good tutorial you can point me to that would be much appreciated.

Thank you,
Tom

J2Kbr, 2020-03-13:
First step is, with the Titan connected to the computer and the wheel at Input-A, open Gtuner IV and go to Device Monitor. Check if the buttons are registered on Device Monitor and create a list of the remapping needed to be done. Please inform the list so I can help you with the remapping.

If you need adjust the wheel sensitivity the script for that can be found here:

https://www.consoletuner.com/wiki/index.php?id=t2:compatibility_table#steering_wheel_controllers

tom10mount, 2020-03-13:
Hi,
The T2 is not recognizing the wheel, I do see that USB A has a HID [IN] for that port but it is not recognizing any buttons.

Thanks,
Tom

https://www.consoletuner.com/wiki/index.php?id=t2:compatibility_table#steering_wheel_controllers

J2Kbr, 2020-03-15:
It is expected the Device Monitor to show: `HID:G [IN]`

I am reopening this issue to further investigate. thank  you.

J2Kbr, 2020-05-13:
Starting from Titan Two firmware 1.8.1.1, the maximum number of COLLECTION items (nested or un-nested) that can be processed in the report item descriptor has been increased from 12 to 26 in order to support the SideWinder Force Feedback Wheel.

closed 2020-05-13 by J2Kbr

tom10mount, 2020-06-13:
Hello, do you have an update on the release date of the 1.8.1.1 update?
Thank you,
Tom