J2Kbr/TitanTwo #258: T2 -> Ps4pro via Output Protocol 'Bluetooth PS4' , BT looses state

Scachi, 2020-02-13:
I have an issue with using my T2 Output as Bluetooth to my Ps4pro using the Output Protocol 'Bluetooth Playstation 4'.
At first it is working fine.. then after some time (something between 14 and 20 Minutes) my T2 looses its Bluetooth functionality.
The Device Monitor states all Bluetooth connections as Disconnected and the Device Configuration Pane shows 'Bluetooth Module : Initializing'
[image]

I have two T2s.. both are doing the same thing so it doesn't look like hardware defect.

Forum Post: https://www.consoletuner.com/forum/viewtopic.php?f=27&t=14751

Scachi, 2020-02-13:
It happens without any script loaded too. Sometimes it shows an  'E4' in the T2 display.

Scachi, 2020-02-13:
This seems to happen only with The Division 2 for me at the moment.
I can play Grip or Ghost Recon Breakpoint for hours without this issue.
I noticed, when starting The Division 2 after doing login into a player char and the map has loaded the T2 led changes to red. Pressing PS button, bringing me back to the playstation menu, will show the LED blue again, switching back to the game and it is red again.

The T2 disconnect happens playing The Division 2 with Bluetooth Output with the Device Lightbar set to "Player Indicator" and it also happens with it set to "Script/Gamepack".
Perhaps The Division 2 uses some specific LED control that causes that problem for the T2 ?

J2Kbr, 2020-02-22:
In the next FW update the Titan Two will report to Gtuner IV any error occurred with Bluetooth connections, this will -hopefully- help identify the cause of what you described.

J2Kbr, 2020-02-26:
The 1.07 Beta FW is now available:

https://www.consoletuner.com/forum/viewtopic.php?f=2&t=14879&p=96076&hilit=Bluetooth#p96076

Please test again with it to check if any debug message is printed in the Gtuner's output panel.

Scachi, 2020-02-28:
Playing "Grid" (not Grip anymore) , the disconnection happened again:
'''
Connect: Titan Two New connected (Internal)
Bluetooth Module: Connection error [ evt 16, err 6, info  ]
Bluetooth Module: Device disconnection [ evt 5, err 0, info 010016 ]
Bluetooth Module: Device disconnection [ evt 5, err 0, info 020016 ]
'''

J2Kbr, 2020-02-28:
Thank you. I believe to have found the cause for this error. The fix will be present in the next 1.07 beta release.

Scachi, 2020-03-04:
There seems to be happened something ..
with the firmware 1.6.3.3 I can use the USB automatic protocol without any issues.

With the  1.7.1.15 it happens that after the boot of my ps4 console I can use my controller for a few minutes, when the connection is lost.
This happens for both, USB and Bluetooth.
I don't get any output in the Output Panel when the controller isn't responding anymore.
Both LED (output and input) stay green and the device configuration pane still reports the bluetooth module as working. So it is different than before but worse as the usb connection now resets after the first minutes sometimes.

I'll update back to the 1.7.1.15 release and check if the device monitor still reports the input (I did forgot to check that)

Scachi, 2020-03-04:
Okay, back at firmware 1.7.1.15, it happened again.
The Device Monitor still reports Inputs and outputs , but no input it getting through to the console..the character isn't moving anymore, can switch to ps4 menu ..nothing works.
But I can still switch the T2 slots with the button combinations.

Device Monitor middle section reports:
[image]

J2Kbr, 2020-03-04:
> With the 1.7.1.15 it happens that after the boot of my ps4 console I can use my controller for a few minutes, when the connection is lost (but I get no disconnect message on the PS4 itself).
> This happens for both, USB and Bluetooth.
I am trying to reproduce this. It only happens with the controller over Bluetooth? or it also happens with the controller wired? (for the case the Titan Two is connected to PS4 via USB cable: USB automatic output protocol).

Thanks.

Scachi, 2020-03-04:
> I am trying to reproduce this. It only happens with the controller over Bluetooth? or it also happens with the controller wired? (for the case the Titan Two is connected to PS4 via USB cable: USB [...]

It happens with bluetooth and usb automatic.
It only happens after a fresh start of the ps4 plus T2.
switch everything off (ps4 & t2). I turn on the T2 first via powersupply. Then my PS4 via its power switch. Then I connect my ds4 controller wireless to the T2, after a few minutes the controller stops to do anything on my ps4.

I'll try to reproduce this tomorrow and try to check if the "turn on" order does matter or not and if the disconnect happens in some specific time frame.

J2Kbr, 2020-03-05:
Thanks, by "_after few minutes_"  it would be something around 7 minutes? If yes, probably is related with the PS4 authentication. Thanks.

Scachi, 2020-03-05:
Yeah, I think that is something around 7 minutes.
I tried to reproduce it but can't figure it out.
Perhaps I did switch everything on first and connected the controller.
Then I did the firmware update, does that reset the authentication ?
(But the T2 display never displayed "AU")
What happens with the valid authentication lease in this case ?
Is the reconnected T2 with the reconnected ds4 controller automatically replacing the previous authentication with this new one ?

Thinking about it the same issue might have happened when trying the bluetooth output.
What happens when I shut down one controller and connect another then ?
Is the authentication lease refreshed with the new connection automatically ?
Is the T2 meant to automatically replace/refresh the auth connection when the old one times out but a new controller is connected ?

Could this be what happened ? :
I used the usb automatic first, with controller via bluetooth.
I then switched the ouput protocol to Bluetooth and reconnected my controller to the T2, then the T2 to the console (via press on controller PS button).
This will work for a few minutes as like the old authentication is still in place, then this old one times out. But the new bluetooth one isn't replacing the usb one when it times out and the T2 isn't showing AU ?
I had to restart my T2 in order to get it working again.

Not sure if that makes any sense.

J2Kbr, 2020-03-05:
Makes sense. The unique ID of the controller is used for the PS4 authentication procedures. The connection needs to be reset if the authenticating controller is changed. The Titan Two does that automatically for higher level pass-thru modes (`none` -> `partial`|`full`, or `partial` -> `full`). If the removed controller is already in full pass-thru mode, the connection is not reset (assuming was just an non-intentional or accidental removal).

closed 2020-03-05 by J2Kbr