J2Kbr/TitanTwo #260: Titan Two bluetooth connection to Nintendo Switch not working

darootler, 2020-02-29:
**Software/Firmware/Hardware used:**
Nintendo Switch: "3.9"
Gtuner IV Beta: 1.7.1.20
Titan Two: 

**Description of problem:**
I am able to pair the Nintendo Switch Pro Controller to the Titan Two device, but i am not able to pair the Titan Two device to the Nintendo Switch. Here is the output of output panel when trying to establish the bluetooth connection:

`
Bluetooth Module: Connection error [ evt 54, err 5, info 5f62bb5c4198 ]
Bluetooth Module: Connection error [ evt 6, err 34, info 0200 ]
Bluetooth Module: Device disconnection [ evt 5, err 0, info 020022 ]
`

If you need more information please let me know.

Regards
Richard

J2Kbr, 2020-03-01:
The first error is the cause of the problem and related to authentication failure. It indicates that authentication failed due to incorrect results in authentication procedure, because an invalid Link Key.

Possible the pairing procedure with the Nintendo Switch console is not succeeding, therefore not generating the Link Key needed for the Bluetooth connection.

In other words: the issue is not Bluetooth related, but with the pairing procedure performed over USB connection.

I will be investigating if there is any difference in the USB pairing procedure between different Nintendo Switch models.

darootler, 2020-03-01:
Okay, let me know if i can do something.

Regards
Richard

J2Kbr, 2020-03-02:
I believe to have found the issue. It is already fixed and will be available in the next FW beta release. Thank you.

darootler, 2020-03-02:
> I believe to have found the issue. It is already fixed and will be available in the next FW beta release. Thank you.

Sounds great, is there an expected release date?

Regards
Richard

J2Kbr, 2020-03-02:
I expect release the beta revision by Wednesday.

closed 2020-03-02 by J2Kbr

darootler, 2020-03-04:
I just updated the Titan Two device to firmware 1.7.1.15, disconnected all controllers from the Nintendo Switch and tried repairing them.

Received the following errors:

```
Bluetooth Module: Connection error [ evt 54, err 5, info 5f62bb5c4198 ]
Bluetooth Module: Connection error [ evt 6, err 34, info 0200 ]
Bluetooth Module: Device disconnection [ evt 5, err 0, info 020022 ]
```

Regards
Richard

J2Kbr, 2020-03-04:
The revision with the fix was not yet released. I am still working in some final improvements.

darootler, 2020-03-11:
I just tested with firmware 1.7.2.0 and received the following error::

{code}
Bluetooth Module: Device disconnection [ evt 5, err 0, info 020000 ]
Bluetooth Module: Connection error [ evt 3, err 4, info 02005f62bb5c41980100 ]
{code}

Should this firmware already include the fix?

Regards
Richard

J2Kbr, 2020-03-11:
This is a different error (timeout). The Titan Two initiated the connection procedures however did not received response from the other device (Nintendo Switch). Did you redid the pairing after update the firmware? (including  disconnecting all controllers from Nintendo Switch). Thanks.