J2Kbr/TitanTwo #262: Moonlight and Parsec virtual controller unique interactions

Seishirou101, 2020-03-27:
After updating to 1.7.3.0 firmware per #247 , (I'm assuming 1.07 is a typo) the issue still exists, the behavior is exactly the same as the video but the USB C is no longer present, as stated by the bug that was fixed with firmware 1.7.0. Parsec has also changed how their controller emulation is done and mirrors moonlight's controller emulation behavior as seen by the past video. See attached screenshot https://puu.sh/FpNJy/b322da5198.png

Seishirou101, 2020-04-06:
As of 4/6, Parsec has fixed the issue with their new update, moonlight is still exhibiting the behavior.

J2Kbr, 2020-05-08:
Correct, the problem with Parsec, and very possible the same with Moonlight, was related with the VID:PID of virtual controller.
 
Gtuner IV explicit checks the VID and HID Usage values to determine how to process the controller input data.
 
If `VID == 0x045E` (Microsoft) and `HID Usage == 0x05` (Gamepad) then the input data is processed as being from a Xbox controller (360 or XB1). Otherwise, the input data is processed as being from a generic HID gaming device. _Side note: Xbox controllers don't have HID descriptors and are not HID compliant, therefore can't be handled by HID parsers._
 
This check is needed because Gtuner IV utilizes the `Raw Input API` to capture the keyboard, mouse and HID gaming devices, not the `Xinput API`. The `Raw Input API` provides access to the raw data from the input device, which requires from Gtuner IV perform any pre-processing needed that would normally be done by a high level API, such as `Xinput`. This extra work is worth, as we can support a wider variety of input devices, as well as forward the keyboard and mouse inputs to the Titan Two simulating as these devices as being direct connected to the Titan Two.

There are, however, an easy workaround which is create a PS4 virtual controller to map the inputs from Moonlight using [reWASD](https://www.rewasd.com).

closed 2020-05-08 by J2Kbr