J2Kbr/TitanTwo #264: Harmonix Guitar Controller for Xbox 360 - add to recognized devices

voltago, 2020-04-09:
Hi,
please include support for the Harmonix Guitar Controller (Wired) for Xbox 360. The ability to script some of the more difficult solos in the Rock Band and Guitar Hero games would be awesome.
Thank you!

```
[USB DESCRIPTOR]
12 01 00 02 FF FF FF 08 AD 1B 02 00 03 01 01 02 03 01 09 02 99 00 04 01 00 A0 FA 09 04 00 00 02 FF 5D 01 00 11 21 10 01 06 25 81 14 03 03 03 04 13 02 08 03 03 07 05 81 03 20 00 04 07 05 02 03 20 00 08 09 04 01 00 04 FF 5D 03 00 1B 21 00 01 01 01 83 40 01 04 20 16 85 00 00 00 00 00 00 16 05 00 00 00 00 00 00 07 05 83 03 20 00 02 07 05 04 03 20 00 04 07 05 85 03 20 00 40 07 05 05 03 20 00 10 09 04 02 00 01 FF 5D 02 00 09 21 00 01 01 22 86 07 00 07 05 86 03 20 00 10 09 04 03 00 00 FF FD 13 04 06 41 00 01 01 03
 
[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 1E
1E 03 48 00 61 00 72 00 6D 00 6F 00 6E 00 69 00 78 00 20 00 4D 00 75 00 73 00 69 00 63 00
 
[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 3A
3A 03 48 00 61 00 72 00 6D 00 6F 00 6E 00 69 00 78 00 20 00 47 00 75 00 69 00 74 00 61 00 72 00 20 00 66 00 6F 00 72 00 20 00 58 00 62 00 6F 00 78 00 20 00 33 00 36 00 30 00
 
[STRING 3 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 03 04 09 00 12
12 03 30 00 30 00 34 00 33 00 39 00 38 00 31 00 34 00 
```

J2Kbr, 2020-05-13:
Thank you for providing the USB descriptors (below decoded for reference).

After checking, this controller should be already supported by the Titan Two via Xbox 360 input driver.

After connecting this controller on the Titan Two, does the status LED of the input port turns green?

At Gtuner's Device Monitor, the status information of the port the controller is plugged shows "`Xbox 360 [IN]`"?

By pressing the controller buttons, is it possible to see activity in the corresponding inputs on Device Monitor?

Thanks.

```
[USB DESCRIPTOR]
12           bLength: 18
01           bDescriptorType: 1 - DEVICE
00 02        bcdUSB: 2.00
FF           bDeviceClass: 255 (FF)
FF           bDeviceSubClass: 255 (FF)
FF           bDeviceProtocol: 255 (FF)
08           bMaxPacketSize0: 8
AD 1B        idVendor: 1BAD
02 00        idProduct: 0002
03 01        bcdDevice: 1.03
01           iManufacturer: 1
02           iProduct: 2
03           iSerialNumber: 3
01           bNumConfigurations: 1

09           bLength: 9
02           bDescriptorType: 2 - CONFIGURATION
99 00        wTotalLength: 153
04           bNumInterfaces: 4
01           bConfigurationValue: 1
00           iConfiguration: 0
A0           bmAttributes: 160 (Bus Powered, Remote Wakeup)
FA           MaxPower: 500mA

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
00             bInterfaceNumber: 0
00             bAlternateSetting: 0
02             bNumEndpoints: 2
FF             bInterfaceClass: 255 (FF)
5D             bInterfaceSubClass: 93 (5D)
01             bInterfaceProtocol: 1 (01)
00             iInterface: 0

11               bLength: 17
21               bDescriptorType: 33 - HID
10 01            bcdHID: 1.10
06               bCountryCode: 6
25               bNumDescriptors: 37
81               bDescriptorType: 129
14 03            wDescriptorLength: 788
03               bDescriptorType: 3
03 04            wDescriptorLength: 1027
13               bDescriptorType: 19
02 08            wDescriptorLength: 2050
03               bDescriptorType: 3
03 07            wDescriptorLength: 1795
05               bDescriptorType: 5
81 03            wDescriptorLength: 897
20               bDescriptorType: 32
00 04            wDescriptorLength: 1024
07               bDescriptorType: 7
05 02            wDescriptorLength: 517
03               bDescriptorType: 3
20 00            wDescriptorLength: 32
08               bDescriptorType: 8
[... 206 lines cut ...]
20 00            wMaxPacketSize: 32
10               bInterval: 16

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
03             bInterfaceNumber: 3
00             bAlternateSetting: 0
00             bNumEndpoints: 0
FF             bInterfaceClass: 255 (FF)
FD             bInterfaceSubClass: 253 (FD)
13             bInterfaceProtocol: 19 (13)
04             iInterface: 4

06               bLength: 6
41               bDescriptorType: 65 - UNKNOW (06 41 00 01 01 03)

[STRING 1 DESCRIPTOR]
Harmonix Music

[STRING 2 DESCRIPTOR]
Harmonix Guitar for Xbox 360

[STRING 3 DESCRIPTOR]
00439814
```

closed 2020-05-13 by J2Kbr

voltago, 2020-10-09:
Thanks for your reply! The guitar controller is recognised in Gtuner and the button presses are visible in the Device Monitor. Everything is fine. **But**: In the game Guitar Hero, the guitar controller is only recognised as a Standard X360 Controller and thus not working. The game only works with a guitar, not a standard controller. 

I'm not sure how the game makes sure a guitar is connected, probably by ProductID/VendorID? Could you please add these IDs?
Would you please reopen this issue? 

Thank you very much!

_Background: I don't know if you're familiar with Guitar Hero, but I'd like to use computer vision to recognise the notes on screen an have them played automatically by T2._