J2Kbr/TitanTwo #268: Feature request: Full button range for HOTAS

Pr3d8or, 2020-06-17:
As requested in [Support for additional buttons](https://www.consoletuner.com/forum/viewtopic.php?f=25&t=11404), it would be good to have direct access to the hid reports of connected controller, so there will be possible access to all inputs.

Desired HOTAS:
- Thrustmaster Warthog
- Logitech (Saitek) X52

thievery80, 2020-10-25:
Hi there!
i am a senior member on an Elite Dangerous Sqadron (160ppl), playing on XBOX.
Almost every day, having a better HOTAS is part of our discussion, most ppl say "If there would be a better-quality solution than the HOTAS One, i´d buy it straight away".
I guess not only our squadron is seeking for better HW support on XBOX/PS4, so i guess, especially once word is spread that there is an easy solution everyone can implement, demands for TITAN can rise quickly.
Therefore, i vote for full button support and GPC listing (y)

makattack, 2020-12-08:
Hello, I am new to GTuner, and happen to have an XBox compatible HOTAS controller (Thrustmaster T-Flight HOTAS One) that works for XBox or PC. I have captured the HID descriptors in "XBox" mode in case it might help with these requests:

`[USB DESCRIPTOR]
12 01 00 02 FF FF FF 40 4F 04 8C B6 00 01 01 02 03 01 09 02 20 00 01 01 00 A0 32 09 04 00 00 02 FF 47 D0 00 07 05 81 03 40 00 04 07 05 01 03 40 00 04 

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 1A 
1A 03 54 00 68 00 72 00 75 00 73 00 74 00 6D 00 61 00 73 00 74 00 65 00 72 00 

[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 26 
26 03 54 00 2E 00 46 00 6C 00 69 00 67 00 68 00 74 00 20 00 48 00 6F 00 74 00 61 00 73 00 20 00 4F 00 6E 00 65 00 

[STRING 3 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 03 04 09 00 22 
22 03 30 00 30 00 30 00 30 00 41 00 30 00 38 00 39 00 44 00 31 00 35 00 34 00 38 00 43 00 31 00 33 00 
`

J2Kbr, 2021-03-16:
HID configuration file for X52 PRO was created and an updated HID configuration file for regular X52. Both will be included into the next Gtuner IV release. Thanks.

closed 2021-03-16 by J2Kbr