J2Kbr/TitanTwo #273: Feature Request: add function to power off the T2 itself

Scachi, 2020-09-14:
When using the bleutooth ps4 output protocol the T2 isn't able to shutdown when the console powers down.
At the moment I am using a small script to turn my controller of but I couldn't find a command to turn of the T2 itself.
I am looking for a way to power down the T2 like the T2 does when the checkbox "Power off the device with the console" is checked. As this checkbox isn't working for me because of bluetooth and external power supply I want to be able to poweroff the T2 by code.

My current code:
`#pragma METAINFO("PortsOff", 1, 0, "Scachi")

#define DWRITE_DI       // disable function DWriteInt(int32 Num) (number to char conversion output)
#define DWRITE_DF       // disable function DWriteFix(fix32 Num) (fix32 numer to char conversion output)
#include "DWrite.gph"

bool off;

main {
    if (check_active(BUTTON_2,1000) && !off) {
        foff();
    }
    if (event_release(BUTTON_2)) off = FALSE;
}

void foff() {
    DWriteTxt("Off");
    off = TRUE;
    if(port_status(PORT_USB_A, NULL, NULL)) port_turnoff(PORT_USB_A);
    if(port_status(PORT_USB_B, NULL, NULL)) port_turnoff(PORT_USB_B);
    if(port_status(PORT_USB_C, NULL, NULL)) port_turnoff(PORT_USB_C);
    if(port_status(PORT_USB_D, NULL, NULL)) port_turnoff(PORT_USB_D);
    
    if(port_status(PORT_BT_A, NULL, NULL)) port_turnoff(PORT_BT_A);
    if(port_status(PORT_BT_B, NULL, NULL)) port_turnoff(PORT_BT_B);
}
`

J2Kbr, 2020-09-19:
Update 1.08 adds a new function `device_poweroff()` for programmatically power off the device.

closed 2020-09-19 by J2Kbr

Scachi, 2020-09-22:
Works perfectly.
Thank you for the fast addition.