J2Kbr/TitanTwo #276: Computer Vision - gcv_ready() stop detecting updates

Scachi, 2020-09-24:
Computer Vision - gcv_ready() stop detecting updates running the current beta FW and gtuner IV 1.8.1.9

We had discussed this some time ago via PM and I don't have the original description anymore.

You reply was:
"After reviewing the firmware code that handles the data received from GCV script I found a bug, which may be the cause of the issue you brought to my attention. I have already fixed and will be available for the next 1.08 beta revision release.
"

Sadly I still have this problem when running computer vision with output protocol Bluetooth PS4.
The .gpc script stops receiving the gcvdata / returned data by the .py script.
Switching back to USB Protocol everything works fine without any changes to the scripts.

I'll try to reproduce it via minimal scripts and will send more information then.

J2Kbr, 2020-10-15:
Could the issue be with the Bluetooth connection with PS4? Like, for some reason, the device stops sending packets to the console. I ask because the output protocol have no connection with the GPC/GCV script processing.

When the issue occurs, please check if the GCV data are being registered by the script using the Device Monitor. If yes, means the issue is with the BT communication with PS4 console.

Thanks.

J2Kbr, 2020-10-31:
Hi Scachi, please comment if you eventually have more information. Thanks.

closed 2020-10-31 by J2Kbr