J2Kbr/TitanTwo #277: Feature Request: Xbox One Drums and Guitar output modes

JesusArmy, 2020-10-18:
Today legacy (x360) drums and guitar connected through Titan Two on Xbox One are recognized as gamepads by Games (Rock Band 4) and in consequence cannot be used to play but only to navigate in game and console menus. Might only be a minor change to add in the device descriptor output for the console do detect as guitar/drum/piano ?
Thanks!

voltago, 2020-10-20:
I have already opened a similar issue, specific to the Harmonix Fender Stratocaster X360 guitar controller (issue #264) and provided USB desscriptors. So I strongly support your feature request and hope to see it implemented soon.
Thanks!

MizzleMonster, 2022-05-07:
I would also love to see this implemented. There are other options I could go with, but having T2 work would be cheaper and much less of a headache!