J2Kbr/TitanTwo #286: Recurring pairing issues with Xbox One wireless adapter to pc through Titan 2

misterpyrrhuloxia, 2021-01-10:
Here's my setup:
- Xbox Elite Series 2 controller > Xbox One Wireless Adapter (Gen 2) > Titan Two > PC
- I also have a second Xbox One Wireless Adapter (Gen 1) that I use for my Steelseries Arctis 9X headset

At least a few times a week, I will run into this issue:
• Xbox One Wireless Adapter isn't recognized correctly by Windows, which causes the Titan Two's Output LED to go red

It's incredibly frustrating and [usually] time-consuming to try to resolve. Most of the time, the only way I can solve the issue is by entirely removing both of my Xbox One Wireless Adapters from the Windows Device Manager and then re-enrolling them. But sometimes, even _that_ gives me issues because I think that Windows gets confused with the Titan Two and having two Xbox wireless adapters. I've basically been bashing my head against for wall for several months now and have no clue what else to do. I have already tried re-formatting my sdcard in Gtuner and re-configuring it. I've already re-installed the latest firmware (both stable and beta builds) and nothing has permanently solved anything for me.

cwiggy78, 2021-01-20:
Hi,

For what its worth I have both wireless adaptor generations too and only run one in my T2 setup but come across this problem every now and again too.  I initially thought my gen 1 wireless adaptor was the culprit so bought the gen 2 but still sporadically get the exact described issue.  It really is a pain trying to unplug and resolve too.  Sometimes needing the whole GTuner IV pair process to be done to re activate connection.

J2Kbr, 2021-03-11:
This issue was fixed with Titan Two FW update 1.10.3.4 (released on Feb 4th, 2021. thanks.

closed 2021-03-11 by J2Kbr