J2Kbr/TitanTwo #288: We need a APIset_mouse to use float

lVentus, 2021-01-14:


Scachi, 2021-01-18:
@J2kbr Questions about the USB HID implementation :
Is it possible to adjust the mouse dpi rate of the simulated mouse ?
At which speed or rate or dpi is the mouse_passthru working ? 
You have mentioned we need to limit the report rate via a combo to not overload the usb connection, how do you avoid that via the passthru function ?
It would be good to be able to use the exact same rate as the mouse_passthru is working for simulating a mouse output via code. Also being able to adjust the dpi rate of that simulated mouse is a must have for being able to get smooth mouse movement output to any game.
Think about using gamepad to mouse output via script for games that require smooth controls like fps shooter. So we need to be able to adjust the effective output value range somehow to match the games required mouse data.

lVentus, 2021-01-18:
> @J2Kbr Questions about the USB HID implementation :
> Is it possible to adjust the mouse dpi rate of the simulated mouse ?
> At which speed or rate or dpi is the mouse_passthru working ?
> You have mentioned we need to limit the report rate via a combo to not overload the usb connection, how do you avoid that via the passthru function ?
> Using higher wait times will result in choppy movement for some games where we can't tune down the games sensitivity enough. Example:
> 
> ```
> #include <mouse.gph>
> 
> main {
>     key_passthru();
>     combo_run(cMove);
> }
> 
> combo cMove {
>     mouse_set(MOUSE_X,0);
>     mouse_set(MOUSE_Y,1);
>     wait(0);
>     wait(6);
> }
> ```
> 
> It would be good to be able to use the exact same rate as the mouse_passthru is working for simulating a mouse output via code. Also being able to adjust the dpi rate of that simulated mouse is a must have for being able to get smooth mouse movement output to any game.
> Think about using gamepad to mouse output via script for games that require smooth controls like fps shooter. So we need to be able to adjust the effective output value range somehow to match the games required mouse data.

Yes，After the discussion in discord,we think we just need a method to set the virtual mouse DPI. We don't actully need a mouse_set of fix32 version. Int32 range of titan2 is 0-6553500. It's accuracy is enough if I can use whole range of int32.
But the default DPI of virtual mouse is too high. When I use mouse_set(mouse_x, 1); and set the sensitivity in game to the lowest. The camera still move too fast.

v-jacob, 2021-01-19:
Yes, this is something I've been having issues as well. I'm trying to create an anti-recoil script for apex legends on PC with mouse passthrough, but mouse_set value of 1 is still too much pull down. Setting the in-game sensitivity low helps but then it's too slow to play.

Scachi, 2021-01-19:
This github issue is more important for direct mouse control, like gamepad to mouse than applying anti recoil.

For anti recoil you can work around that limitation most of the time by using a higher the mouse speed value in that script: https://www.consoletuner.com/wiki/index.php?id=t2:gpc_scripting:usb_hid#anti_recoil

v-jacob, 2021-01-23:
Setting a higher value for mouse speed results in choppy mouse movements. I'll give the usb hid warzone script a try.

Scachi, 2021-01-23:
> Setting a higher value for mouse speed results in choppy mouse movements. I'll give the usb hid warzone script a try.

I also added a working example code with decimal support to the wiki:
https://www.consoletuner.com/wiki/index.php?id=t2:gpc_scripting:usb_hid#anti_recoil_decimal

lVentus, 2021-01-23:
> > Setting a higher value for mouse speed results in choppy mouse movements. I'll give the usb hid warzone script a try.
> 
> I also added a working example code with decimal support to the wiki:
> https://www.consoletuner.com/wiki/index.php?id=t2:gpc_scripting:usb_hid#anti_recoil_decimal

Hello,I have solve this problem by myself.

J2Kbr, 2021-04-01:
The mouse DPI is not something that is configured, for example, on the device USB descriptors.

In practice a high DPI mouse just sends higher values (e.g. `mouse_set(MOUSE_X, <mouse_value>);` for the same traveled distance.

Anyway, the newly added function on Gtuner/Titan Two 1.11 (to be released soon), can be used to sync the generated mouse data packets with the output polling rate. This can be used to implement scripts with more precise mouse movement.

More info here: #271

closed 2021-04-01 by J2Kbr