J2Kbr/TitanTwo #289: Alternate remote slot change combination

Scachi, 2021-01-19:
by  cwiggy78 at the forums:
```
Hi,

This has been driving me bananas for a while. I play a lot of COD and FPS in general.

Please @J2Kbr could you place an alternate remote slot combination in the firmware :

Down Slot : LS + Xbox button
UP Slot : RS + Xbox button

It is really tricky to use the existing combination as the controller bumpers are usually tied in game to grenades and such.

Kind regards
cwiggy78
```

I would like to have that option too and it was requested a lot of times already to have an alternate slot changing option.

While it is true that we can  open a menu first before pressing that combination it add more button presses to it to just change the slot.

This alternate button combination as suggested would only invoke a "run" or a "melee" attack in most games and nothing harmful like throwing a grenade as the current combination is doing.

cwiggy78, 2021-01-19:
Hi,

Thank you for adding this request. I believe it will be a game changer for FPS users and will definitely improve the speed in which a slot change can be made on the fly.  This just improves upon an already fantastic product.

Regards

cwiggy78

J2Kbr, 2021-04-01:
Alternate button combination for memory slot selection using the controller added to firmware 1.11. Thank you.

closed 2021-04-01 by J2Kbr