J2Kbr/TitanTwo #294: add HID for "Saitek/Logitech X52" / compare to HID for "Saitek/Logitech X52 pro"

thievery80, 2021-03-15:
As the X52 (next to the "X52 pro") is a very common Joystick for Flight/Space-Sim, we as endusers with X52 want to use all Buttons/Sliders/Wheels.
As for XBOX OUT, there is some workaround, but still we miss some buttons, wheels, microstick; for PS4 (and for PC) OUT, there is another HID in use where some of those buttons work, but others not.
As endusers, we want to have an universal, fully functional HID for either "X52" or "X52 pro" no matter to wich destination the OUTPUT (PS/XBOX/PC/...) is connected.

Please find the HID descriptors for my "Saitek X52), as well as a file for a brandnew "Logitech X52" attached.
[Saitek_X52_HID.txt](https://github.com/J2Kbr/TitanTwo/files/6140921/Saitek_X52_HID.txt)
[Logitech_X52_HID.txt](https://github.com/J2Kbr/TitanTwo/files/6140930/Logitech_X52_HID.txt)

J2Kbr, 2021-03-16:
HID configuration file for X52 PROo was created and an updated HID configuration file for regular X52. Both will be included into the next Gtuner IV release. Thanks.

closed 2021-03-16 by J2Kbr

thievery80, 2021-03-16:
WOW! you literally have no idea how much this means to me (and the guys i am working with)!!!!

Many Thanks, keep up the good work!